Eomys/pyleecan #581: From Jan 1, 2023, ndim will require a Sigma license to run.

crowdnidhi-design, 2022-12-20:
Dear All,

Well, what is the solution for the above issue.
When I launch pyleecan I get the above message.

BonneelP, 2022-12-22:
Hello,

This message is linked to a change in quadpy/ndim licenses. We didn't find any information about what the Sigma license will be/cost but quadpy is only used in one non-mandatory methods Methods\Mesh\FPGNTri\get_gauss_points.py:

```python
def get_gauss_points(self):
    """Return the gauss points and weights for Triangle3 cell"""

    nb_gauss_points = self.nb_gauss_point
    scheme = quadpy.tn.grundmann_moeller(2, nb_gauss_points)
    gauss_pts = scheme.points
    weights = scheme.weights

    return gauss_pts, weights, nb_gauss_points
```

We will see what happens next year but we should be able to remove the quadpy requirements. 

Best regards,
Pierre

crowdnidhi-design, 2022-12-30:
Thank you, Pierre.

schnib, 2023-01-03:
Hey,
importing quadpy is now not possible anymore without a sigma license.
As it's being imported when a Simulation object is created, it isn't possible to run simulations anymore.

BonneelP, 2023-01-03:
Hello, 
Yes we have the same issue. We have something in pyleecan that skip an import when a package is not used and we thought that it would work. But we will just remove the quadpy dependency completely and create a new pypi release.

Best regards,
Pierre

BonneelP, 2023-01-03:
New pyleecan 1.4.2 release is available on pypi. It should solve the issue (quadpy is no longer imported, but we recommend to uninstall it anyway).

RaphaelPile, 2023-01-11:
Hello there ! 
I can do the modification in the code to be independent from Quadpy if necessary.