Eomys/pyleecan #583: Can we integrate Pyleecan with Scilab/XCOS?.

crowdnidhi-design, 2022-12-30:
Dear Pierre,

Just curious do you have MTPA, FOC, or DTC options/libraries /Algo available as a functionality/feature?
Can we integrate Pyleecan with Scilab/XCOS?.
If yes please share the procedure to integrate with a few examples...

BonneelP, 2023-01-03:
Hello,

Regarding MTPA there are few functions and tests within pyleecan that will be improved by #558. Otherwise we have few scripts/Tests that uses EEC (Electrical Equivalent circuit) but most of the time simulation are sinusoidal in pyleecan. The exception is PWM that is available for PMSM so harmonic current is also possible (and enforced harmonic current shouldn't be too hard to implement).
There is no coupling with Scilab/XCOS but it could be interesting to investigate that. If you have some time to work on the subject I should be able to guide you.

Best regards,
Pierre

crowdnidhi-design, 2023-01-03:
Dear Pierre,

Sure. Let me set up some modules with Scilab/XCOS and get back to you.

Any update/ feedback on FOC, DTC , etc .
Further, do you have support to generate SVM ( 3 or multistage)?.