Eomys/pyleecan #584: Request for feedback - Holes in-between 2 poles in rotor

crowdnidhi-design, 2023-01-02:
Dear All,

I am curious to understand how to create a hole between two poles( in a rotor). When we import DXF files it is creating a line and femm analysis breaks.

BonneelP, 2023-01-03:
Hello,

If I understand correctly you need to set your HoleUD as an axial ventilation duct. 
- rotor.hole are holes for pole (with/without magnets) so by definition they are not between two poles
- rotor.axial_vent are empty holes that can be placed anywhere
Can you share your topology ?

Best regards,
Pierre

crowdnidhi-design, 2023-01-03:
[image]

crowdnidhi-design, 2023-01-03:
We are trying to create the above. Can you find the( elliptical) hole between the poles near the shaft?.
If we import the dxf files( one pole slice) and create it we are getting a line in the center and the analysis fails. 
Do you expect us to import the complete rotor dxf ( all poles).

One more quick question. Can we import the complete Rotor design rather than one pole piece?.....

BonneelP, 2023-01-04:
Regarding your other question, you can't import the full rotor, LamHole is meant to import several set of holes that will be duplicated 2*p times. There is LamHoleNS that enables to define a different set of holes for the North and the South (each duplicated p times). You could take inspiration from LamHoleNS to create LamHoleAll with all_holes properties to store in order each 2*p holes.