Eomys/pyleecan #585: This version of orthopy is no longer supported. Please update.

crowdnidhi-design, 2023-01-03:
Dear All,

When I launch Pyleecan from the command prompt I get the above message and I cant launch the GUI.

crowdnidhi-design, 2023-01-03:
Installing collected packages: orthopy
  Attempting uninstall: orthopy
    Found existing installation: orthopy 0.8.92
    Uninstalling orthopy-0.8.92:
      Successfully uninstalled orthopy-0.8.92
Successfully installed orthopy-0.8.93

C:\Users\Haris>python -m pyleecan
Warning: Unable to find valid license in
- environment variable SIGMA_LICENSE_KEY (env var not found), or
- config file C:\Users\Haris\AppData\Local\sigma\sigma.toml (file not found)

Enter new license key for sigma:

Can someone help?

BonneelP, 2023-01-03:
Hello,
I think this is linked to issue #581, can you uninstall quadpy ?
I will create a new release.

crowdnidhi-design, 2023-01-03:
and please update your website. No information on the latest release and how to install it yet.

crowdnidhi-design, 2023-01-03:
It seems to be working now. Thank you for ur new release. I hope u can update ur website. 
Happy New Year.

BonneelP, 2023-01-04:
Hello, 
Which part of the website needs to be updated ? 1.4.2 doesn't change that much the content of the exe and I checked the exe from 1.4.1 is still working on my side. 
Happy New Year,
Pierre

crowdnidhi-design, 2023-01-05:
https://www.pyleecan.org/