Eomys/pyleecan #586: Error while saving results: module 'numpy' has no attribute 'warnings'

crowdnidhi-design, 2023-01-04:
Dear All,

I am getting the above warning after I upgraded to a new version yesterday.

Do you have any idea how to remove this warning pls?

BonneelP, 2023-01-04:
Hello,

Which version of numpy are you using ? I have 1.21.4 on my side.

Best regards,
Pierre

crowdnidhi-design, 2023-01-04:
Name: numpy
Version: 1.24.0
Summary: Fundamental package for array computing in Python
Home-page: https://www.numpy.org
Author: Travis E. Oliphant et al.
Author-email:
License: BSD-3-Clause
Location: c:\users\haris\appdata\local\programs\python\python38\lib\site-packages
Requires:
Required-by: autograd, cma, contourpy, cplot, deap, h5py, imageio, matplotlib, matplotx, meshio, npx, orthopy, pandas, pyDOE2, pyleecan, pymoo, pyqtgraph, pyuff, pyvista, quadpy, SciDataTool, scikit-learn, scipy, swat-em

crowdnidhi-design, 2023-01-04:
One more issue has started, whatever we are trying simulation is giving an error 
[20:16:26] Starting Magnetic module
[20:16:26] Computing Airgap Flux in FEMM
[20:16:28] Error while running simulation:
error: Material properties have not been defined for all regions
[20:17:36] Error while running simulation:
error: Material properties have not been defined for all regions
[20:17:36] Error while saving results:
local variable 'out' referenced before assignment

crowdnidhi-design, 2023-01-04:
Well, it was working for the first time. 
I recreated everything now it is not working. Tried a different design again it does not work.
Well if we save the file in a different location does it create this issue? In FEM I am noticing it is not assigning material properties to Rotor laminate. 
I tried to create it in FEM and then exported the DXF file. Well, it fails again. Is something wrong ?...

BonneelP, 2023-01-04:
All seems to be working on my side with the default machines.
Can you try to run a simulation with the Prius ?
Can you share your machine/simulation/fem file ?

crowdnidhi-design, 2023-01-04:
yes, Prius works. But all that we are trying does not work from the evening...
I am trying a new model let us see if it works

crowdnidhi-design, 2023-01-04:
Dear Pierre,

It seems the DXF file is the issue.
I tried a new model using features in your tool and simulation works.
It fails when I try to import some random design using a DXF import from either FEM or CAD.

I will try in a different way tomorrow.

Good night.

Regards,

crowdnidhi-design, 2023-01-05:
Dear Pierre,

What should be the tolerance we can maintain in CAD drawings?. It seems if we have slits or gaps of less than 50 microns we are getting a variety of errors. Solving one after the other is taking a lot of time but the same if we directly try using FEMM.info software it works like charm. we are trying designs with 30-40 microns i.e. example space between magnets and laminates, etc so that we can look at deviations due to antirust coatings, adhesives, rotor shaft and rotor - shrinkfit ....etc

Your feedback would be much appreciated. 
One more observation. I have noticed sometimes the magnet defaults to Prius magnets. Not sure why?...

example of some errors 

Saving Simu1 to file 'C:/Users/Haris/Downloads/2023_01_05-14h32min09s_FEMM_example\FEMM_example.json'.
[14:32:09] Starting running simulation FEMM_example (machine=example)
[14:32:09] Starting Magnetic module
[14:32:10] Computing Airgap Flux in FEMM
[14:36:50] Error while running simulation:
error: Some regions in the problem have been defined
by more than one block label.  These potentially
problematic regions will appear as selected in
the initial view.
[15:34:31] Error while running simulation:
error: Some regions in the problem have been defined
by more than one block label.  These potentially
problematic regions will appear as selected in
the initial view.
[15:34:31] Error while saving results:
local variable 'out' referenced before assignment
[15:34:32] Error while saving results:
local variable 'out' referenced before assignment
Traceback (most recent call last):
  File "c:\users\haris\downloads\pyleecan-1.4.2.tar\pyleecan-1.4.2\pyleecan-1.4.2\pyleecan\GUI\Dialog\DMachineSetup\SSimu\SSimu.py", line 206, in run
    out.simu.machine.plot(
UnboundLocalError: local variable 'out' referenced before assignment

crowdnidhi-design, 2023-01-05:
Dear Pierre,

Solved the issues. I believe it has to do with tolerance. Was able to run the simulation. But I am still getting the issue wrt numpy.

[20:10:16] Starting running simulation FEMM_example (machine=example)
[20:10:16] Starting Magnetic module
[20:10:16] Computing Airgap Flux in FEMM
[20:22:10] End of simulation FEMM_example
[20:22:10] Error while saving results:
module 'numpy' has no attribute 'warnings'
[20:22:19] Error while saving results:
module 'numpy' has no attribute 'warnings'