Eomys/pyleecan #588: Issues with Datakeeper object in Tutorials

SupaDupaTwo, 2023-01-06:
Hello,
I encounter some errors while trying to understand the tutorials for operating point and Optimization. Initialization of the DataKeeper object fails. Can you spot the issue?

In tuto_Operating_point.ipynb as well as in tuto_Optimization.ipynb
```
TypeError                                 Traceback (most recent call last)
Cell In[4], line 12
      8     # Return the first torque harmonic
      9     return harm_list[1] 
     11 my_obj = [
---> 12     OptiObjective(
     13         name="Maximization of the average torque",
     14         symbol="Tem_av",
     15         unit="N.m",
     16         keeper="lambda output: -abs(output.mag.Tem_av)",  # keeper can be saved
     17     ),
```
`TypeError: 'property' object is not callable`

```
TypeError                                 Traceback (most recent call last)
Cell In[21], line 41
     38 varload.is_reuse_femm_file=True  
     40 # Add I0/Phi0 to the data available at the end of the simulation
---> 41 I0_dk = DataKeeper(
     42     name="Stator current rms amplitude",
     43     symbol="I0",
     44     unit="Arms",
     45     keeper="lambda output: output.elec.OP.get_I0_Phi0()['I0']",
     46 )
```
`TypeError: 'property' object is not callable`

BonneelP, 2023-01-06:
Hello, 
The code looks right to me but we have recently merged a modification on the DataKeeper/Optimization so maybe we missed something. Which version of pyleecan are you using ? Are you running the latest version of the tutorial notebook without modifications ?
Best regards,
Pierre

SupaDupaTwo, 2023-01-06:
I pulled the commit bda513c8 on the master branch
Python 3.10.9
Pyleecan version:1.4.2
SciDataTool version:2.5.0

I did not modify tuto_Optimizaton.ipynb but I did modify tuto_Operating_point.ipynb
-`varload.set_OP_matrix(OP_matrix, "N0", "I0", "Phi0", "Tem")`
+`varload.set_OP_matrix(OP_array, "N0", "I0", "Phi0", "Tem")`

dyf502, 2023-02-14:
Is the problem solved? I have the same problem，I am  running the latest version of the tutorial notebook without modifications.
yleecan version:1.4.2
SciDataTool version:2.5.0

BonneelP, 2023-04-26:
Hello,

Normally this bug is corrected on the master branch of the project. I will try to merge #616 and create a new release this week to solve this issue.

Best regards,
Pierre

faraz4714, 2023-04-26:
hello,
thank you for your reply. 
I got confused is there available any code and datakeeper class that this bug was solved or I should wait for the new release?
if yes it's possible to guide me to access them 

best regards,
faraz

BonneelP, 2023-04-26:
Hello,

You can get the code from Github using method 3: https://pyleecan.org/get.pyleecan.html
You can then use the code that is not released yet and change it if needed. 
If you are not familiar with python or git, then you can wait for the release and you will just need to run pip install again.

Best regards,
Pierre

SupaDupaTwo, 2023-05-02:
I still wonder what python version you are using, some package are quite old. I tried installing pyleecan 1.50 in a new environement and I have error with matplotlib and CedMrnl fork of swat-em (btw, what are the difference with the pip version of swat-em?)

```
     copying lib\matplotlib\mpl-data\fonts\afm\cmr10.afm -> build\lib.win-amd64-cpython-310\matplotlib\mpl-data\fonts\afm
      UPDATING build\lib.win-amd64-cpython-310\matplotlib\_version.py
      set build\lib.win-amd64-cpython-310\matplotlib\_version.py to '3.3.4'
      running build_ext
      error: Failed to download FreeType. Please download one of ['https://downloads.sourceforge.net/project/freetype/freetype2/2.6.1/freetype-2.6.1.tar.gz', 'https://download.savannah.gnu.org/releases/freetype/freetype-2.6.1.tar.gz'] and extract it into build\freetype-2.6.1 at the top-level of the source repository.
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.

  ERROR: Failed building wheel for matplotlib

Failed to build matplotlib
```

```
Cloning https://gitlab.com/CedMrnl/swat-em.git to c:\users\david.buehlmann\appdata\local\temp\pip-req-build-434ia23a
  Running command git clone --filter=blob:none --quiet https://gitlab.com/CedMrnl/swat-em.git 'C:\Users\david.buehlmann\AppData\Local\Temp\pip-req-build-434ia23a'
  fatal: unable to access 'https://gitlab.com/CedMrnl/swat-em.git/': SSL certificate problem: unable to get local issuer certificate
```

BonneelP, 2023-05-02:
Hello,

As discussed in other issues, pyleecan is not (yet) compatible with Python 3.11 so I have added this limitation in pip.
The issue is that as 1.5.0 is flagged as not available for Python 3.11, while using pip it revert back to the previous version which is 1.4.2 (which is not flagged as incompatible with 3.11) 
Regarding the modification for swat-em, we have an opened PR on the project that changes PyQt5 import to qtpy import since we are using PySide2.

Best regards,
Pierre

BonneelP, 2023-05-02:
By the way I'm personally working on Python 3.8.10

faraz4714, 2023-05-02:
Hello,
I downgrade my python to 3.8.10 and now I have pyleecan 1.5 but now there is a strange error. i used pip install pyleecan more than one but there is this error 
```
`%matplotlib inline

# Load the machine
from os.path import join
from pyleecan.Functions.load import load
from pyleecan.definitions import DATA_DIR

IPMSM_A = load(join(DATA_DIR, "Machine", "Toyota_Prius.json"))
# In Jupyter notebook, we set is_show_fig=False to skip call to fig.show() to avoid a warning message
# All plot methods return the corresponding matplotlib figure and axis to further edit the resulting plot
fig, ax = IPMSM_A.plot(is_show_fig=False)  
```

```
`ModuleNotFoundError                       Traceback (most recent call last)
Cell In[3], line 5
      3 # Load the machine
      4 from os.path import join
----> 5 from pyleecan.Functions.load import load
      6 from pyleecan.definitions import DATA_DIR
      8 IPMSM_A = load(join(DATA_DIR, "Machine", "Toyota_Prius.json"))

File ~\AppData\Local\Programs\Python\Python38\Lib\site-packages\shiboken2\files.dir\shibokensupport\__feature__.py:142, in _import(name, *args, **kwargs)
    140     flag = -1
    141 pyside_feature_dict[importing_module] = flag
--> 142 return original_import(name, *args, **kwargs)

File c:\users\faraz\appdata\local\programs\python\python38\lib\site-packages\pyleecan\Functions\load.py:8
      6 from .Load.import_class import import_class
      7 from .Load.load_hdf5 import load_hdf5
----> 8 from .Load.load_json import load_json
      9 from .Load.load_pkl import load_pkl
     10 from .Load.retrocompatibility import convert_init_dict

File ~\AppData\Local\Programs\Python\Python38\Lib\site-packages\shiboken2\files.dir\shibokensupport\__feature__.py:142, in _import(name, *args, **kwargs)
    140     flag = -1
    141 pyside_feature_dict[importing_module] = flag
--> 142 return original_import(name, *args, **kwargs)

File c:\users\faraz\appdata\local\programs\python\python38\lib\site-packages\pyleecan\Functions\Load\load_json.py:4
      2 from json import load as jload
      3 from os.path import isdir, isfile, splitext
----> 4 from ...Generator import PYTHON_DEFAULT_ENCODING
      7 def load_json(file_path):
      8     """Load a json file
      9 
     10     Parameters
   (...)
     20         data of the json file
     21     """

File ~\AppData\Local\Programs\Python\Python38\Lib\site-packages\shiboken2\files.dir\shibokensupport\__feature__.py:142, in _import(name, *args, **kwargs)
    140     flag = -1
    141 pyside_feature_dict[importing_module] = flag
--> 142 return original_import(name, *args, **kwargs)

ModuleNotFoundError: No module named `'pyleecan.Generator'``
```

BonneelP, 2023-05-02:
Hello,

I seems that a file is missing in the pip release. I will check this afternoon and if needed I will release 1.5.1.

Best regards,
Pierre

BonneelP, 2023-05-02:
Hello,

Which version of VTK are you using ?
On my side I have:
vtk 9.0.3
pyvista 0.31.3
pyvistaqt 0.5.0
meshio 4.4.6

Best regards,
Pierre

faraz4714, 2023-05-03:
hello 
thank you for your response 
i set all my libraries as you but I get new error related to numpy 

```
AttributeError                            Traceback (most recent call last)
Cell In[14], line 1
----> 1 out.loss.loss_dict["overall"].plot_mesh()  # overall on full mesh

File c:\users\faraz\appdata\local\programs\python\python38\lib\site-packages\pyleecan\Methods\Output\OutLossModel\plot_mesh.py:13, in plot_mesh(self, group_names)
      1 def plot_mesh(self, group_names=None):
      2     """Plot the losses on the mesh solution
      3 
      4     Parameters
   (...)
     10 
     11     """
---> 13     self.get_mesh_solution().plot_contour(
     14         "freqs=sum",
     15         label=self.name + " loss density [W/m^3]",
     16         group_names=group_names,
     17     )

File c:\users\faraz\appdata\local\programs\python\python38\lib\site-packages\pyleecan\Methods\Mesh\MeshSolution\plot_contour.py:164, in plot_contour(self, label, index, indices, is_surf, is_radial, is_center, clim, field_name, group_names, save_path, itimefreq, is_show_fig, win_title, factor, is_animated, title, p, colormap, *args)
    161         clim[0] = -abs(clim[1])
    162         clim[1] = abs(clim[1])
--> 164 plot_mesh_field(
    165     p,
    166     sargs,
    167     field_name,
    168     clim=clim,
    169     mesh_pv=mesh_pv,
    170     field=field,
    171     colormap=colormap,
    172 )
    174 ###########
    175 # Internal animation (cannot be combined with other plots)
    176 if is_animated:

File c:\users\faraz\appdata\local\programs\python\python38\lib\site-packages\pyleecan\Functions\Plot\Pyvista\plot_mesh_field.py:24, in plot_mesh_field(p, sargs, field_name, clim, mesh_pv, field, phase, colormap)
     13 def plot_mesh_field(
     14     p,
     15     sargs,
   (...)
     21     colormap=COLOR_MAP,
     22 ):
---> 24     mesh_pv[field_name] = real(field * phase)
     25     mesh_field = mesh_pv
     27     p.add_mesh(
     28         mesh_field,
     29         scalars=field_name,
   (...)
     33         scalar_bar_args=sargs,
     34     )

File c:\users\faraz\appdata\local\programs\python\python38\lib\site-packages\pyvista\core\dataset.py:654, in DataSet.__setitem__(self, name, scalars)
    652     self.point_arrays[name] = scalars
    653 elif scalars.shape[0] == self.n_cells:
--> 654     self.cell_arrays[name] = scalars
    655 else:
    656     # Field data must be set explicitly as it could be a point of
    657     # confusion for new users
    658     raise_not_matching(scalars, self)

File c:\users\faraz\appdata\local\programs\python\python38\lib\site-packages\pyvista\core\datasetattributes.py:57, in DataSetAttributes.__setitem__(self, key, value)
     55 def __setitem__(self, key: str, value: np.ndarray):
     56     """Implement setting with the [] operator."""
---> 57     self.append(narray=value, name=key)

File c:\users\faraz\appdata\local\programs\python\python38\lib\site-packages\pyvista\core\datasetattributes.py:247, in DataSetAttributes.append(self, narray, name, deep_copy, active_vectors, active_scalars)
[... 44 lines cut ...]
     85                 vtkConstants.VTK_UNSIGNED_LONG_LONG:numpy.uint64,
     86                 vtkConstants.VTK_ID_TYPE:ID_TYPE_CODE,
     87                 vtkConstants.VTK_FLOAT:numpy.float32,
     88                 vtkConstants.VTK_DOUBLE:numpy.float64}
     89     return _vtk_np

File c:\users\faraz\appdata\local\programs\python\python38\lib\site-packages\numpy\__init__.py:305, in __getattr__(attr)
    300     warnings.warn(
    301         f"In the future `np.{attr}` will be defined as the "
    302         "corresponding NumPy scalar.", FutureWarning, stacklevel=2)
    304 if attr in __former_attrs__:
--> 305     raise AttributeError(__former_attrs__[attr])
    307 # Importing Tester requires importing all of UnitTest which is not a
    308 # cheap import Since it is mainly used in test suits, we lazy import it
    309 # here to save on the order of 10 ms of import time for most users
    310 #
    311 # The previous way Tester was imported also had a side effect of adding
    312 # the full `numpy.testing` namespace
    313 if attr == 'testing':

AttributeError: module 'numpy' has no attribute 'bool'.
`np.bool` was a deprecated alias for the builtin `bool`. To avoid this error in existing code, use `bool` by itself. Doing this will not modify any behavior and is safe. If you specifically wanted the numpy scalar type, use `np.bool_` here.
The aliases was originally deprecated in NumPy 1.20; for more details and guidance see the original release note at:
    https://numpy.org/devdocs/release/1.20.0-notes.html#deprecations
```

SupaDupaTwo, 2023-05-03:
I think with issue comes from Pyvista. I had a similar issue and I had to change the import in Lib/site-packages/pyvista/_vtk.py, vtkExtractEdges is in vtkmodules.vtkFiltersCore now.
In get_cell.py I had also to change the import -> collections.abc import Iterable

BonneelP, 2023-05-04:
Hello,

Thank you for the answer. I'm not sure if understand perfectly your answer. Should we change some import within pyleecan or is it modifications to do in the site_package of VTK ? In the first case do you have the time to create a PR ? I can release 1.5.2 afterwards if needed.
By the way, I'm using numpy 1.23.1

Best regards,
Pierre

SupaDupaTwo, 2023-05-05:
The issue is within pyvista I think. To solve it I changed the site_package of pyvista. A better option should be to upgrade pyvista to a newer version. If it is possible with the other requirements of pyleecan. 
[https://github.com/pyvista/pyvista/pull/2585](https://github.com/pyvista/pyvista/pull/2585).

\pyleecan\Methods\Mesh\MeshMat\get_cell.py had a issue of import of the collections module but I think it is fixed in 1.50
I am running the unit test with the latesta pyvista to see it create new issue

faraz4714, 2023-05-10:
hello,
thank you for your answer I change the versions of the numpy and matplotlib and now work correctly 
Best regards,
Faraz Mirali