Eomys/pyleecan #590: Not able to launch Pyleecan

crowdnidhi-design, 2023-01-06:
Dear All,

Well, everything was working fine till afternoon. 
I installed Anaconda and now I cant launch Pyleecan from the command prompt...
Are these are related to each other ?.
How to resolve the issue please? 

C:\Users\Haris>python -m pyleecan
Traceback (most recent call last):
  File "C:\Users\Haris\AppData\Local\Programs\Python\Python38\lib\runpy.py", line 194, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "C:\Users\Haris\AppData\Local\Programs\Python\Python38\lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "c:\users\haris\downloads\pyleecan-1.4.2.tar\pyleecan-1.4.2\pyleecan-1.4.2\pyleecan\__main__.py", line 2, in <module>
    from .run_GUI import run_GUI
  File "c:\users\haris\downloads\pyleecan-1.4.2.tar\pyleecan-1.4.2\pyleecan-1.4.2\pyleecan\run_GUI.py", line 15, in <module>
    from .GUI.Dialog.DMachineSetup.DMachineSetup import DMachineSetup
  File "C:\Users\Haris\AppData\Local\Programs\Python\Python38\Lib\site-packages\shiboken2\files.dir\shibokensupport\__feature__.py", line 142, in _import
    return original_import(name, *args, **kwargs)
  File "c:\users\haris\downloads\pyleecan-1.4.2.tar\pyleecan-1.4.2\pyleecan-1.4.2\pyleecan\GUI\Dialog\DMachineSetup\DMachineSetup.py", line 11, in <module>
    from ....GUI.Dialog.DMachineSetup.SPreview.SPreview import SPreview
  File "C:\Users\Haris\AppData\Local\Programs\Python\Python38\Lib\site-packages\shiboken2\files.dir\shibokensupport\__feature__.py", line 142, in _import
    return original_import(name, *args, **kwargs)
  File "c:\users\haris\downloads\pyleecan-1.4.2.tar\pyleecan-1.4.2\pyleecan-1.4.2\pyleecan\GUI\Dialog\DMachineSetup\SPreview\SPreview.py", line 8, in <module>
    from .....GUI.Dialog.DMachineSetup.SPreview.Ui_SPreview import Ui_SPreview
  File "C:\Users\Haris\AppData\Local\Programs\Python\Python38\Lib\site-packages\shiboken2\files.dir\shibokensupport\__feature__.py", line 142, in _import
    return original_import(name, *args, **kwargs)
  File "c:\users\haris\downloads\pyleecan-1.4.2.tar\pyleecan-1.4.2\pyleecan-1.4.2\pyleecan\GUI\Dialog\DMachineSetup\SPreview\Ui_SPreview.py", line 12, in <module>
    from .....GUI.Dialog.DMachineSetup.SPreview.WMachineTable.WMachineTable import (
  File "C:\Users\Haris\AppData\Local\Programs\Python\Python38\Lib\site-packages\shiboken2\files.dir\shibokensupport\__feature__.py", line 142, in _import
    return original_import(name, *args, **kwargs)
  File "c:\users\haris\downloads\pyleecan-1.4.2.tar\pyleecan-1.4.2\pyleecan-1.4.2\pyleecan\GUI\Dialog\DMachineSetup\SPreview\WMachineTable\WMachineTable.py", line 8, in <module>
    from ......Classes.InputCurrent import InputCurrent
  File "C:\Users\Haris\AppData\Local\Programs\Python\Python38\Lib\site-packages\shiboken2\files.dir\shibokensupport\__feature__.py", line 142, in _import
    return original_import(name, *args, **kwargs)
  File "c:\users\haris\downloads\pyleecan-1.4.2.tar\pyleecan-1.4.2\pyleecan-1.4.2\pyleecan\Classes\InputCurrent.py", line 21, in <module>
    from ..Methods.Simulation.InputCurrent.gen_input import gen_input
  File "C:\Users\Haris\AppData\Local\Programs\Python\Python38\Lib\site-packages\shiboken2\files.dir\shibokensupport\__feature__.py", line 142, in _import
    return original_import(name, *args, **kwargs)
  File "c:\users\haris\downloads\pyleecan-1.4.2.tar\pyleecan-1.4.2\pyleecan-1.4.2\pyleecan\Methods\Simulation\InputCurrent\gen_input.py", line 6, in <module>
    from ....Classes.ElecLUTdq import ElecLUTdq
  File "C:\Users\Haris\AppData\Local\Programs\Python\Python38\Lib\site-packages\shiboken2\files.dir\shibokensupport\__feature__.py", line 142, in _import
    return original_import(name, *args, **kwargs)
  File "c:\users\haris\downloads\pyleecan-1.4.2.tar\pyleecan-1.4.2\pyleecan-1.4.2\pyleecan\Classes\ElecLUTdq.py", line 21, in <module>
    from ..Methods.Simulation.ElecLUTdq.run import run
  File "C:\Users\Haris\AppData\Local\Programs\Python\Python38\Lib\site-packages\shiboken2\files.dir\shibokensupport\__feature__.py", line 142, in _import
    return original_import(name, *args, **kwargs)
  File "c:\users\haris\downloads\pyleecan-1.4.2.tar\pyleecan-1.4.2\pyleecan-1.4.2\pyleecan\Methods\Simulation\ElecLUTdq\run.py", line 4, in <module>
    from ....Classes.OutLoss import OutLoss
  File "C:\Users\Haris\AppData\Local\Programs\Python\Python38\Lib\site-packages\shiboken2\files.dir\shibokensupport\__feature__.py", line 142, in _import
    return original_import(name, *args, **kwargs)
  File "c:\users\haris\downloads\pyleecan-1.4.2.tar\pyleecan-1.4.2\pyleecan-1.4.2\pyleecan\Classes\OutLoss.py", line 46, in <module>
    from ..Methods.Output.OutLoss.store import store
  File "C:\Users\Haris\AppData\Local\Programs\Python\Python38\Lib\site-packages\shiboken2\files.dir\shibokensupport\__feature__.py", line 142, in _import
    return original_import(name, *args, **kwargs)
  File "c:\users\haris\downloads\pyleecan-1.4.2.tar\pyleecan-1.4.2\pyleecan-1.4.2\pyleecan\Methods\Output\OutLoss\store.py", line 4, in <module>
    from ....Classes.MeshSolution import MeshSolution
  File "C:\Users\Haris\AppData\Local\Programs\Python\Python38\Lib\site-packages\shiboken2\files.dir\shibokensupport\__feature__.py", line 142, in _import
    return original_import(name, *args, **kwargs)
  File "c:\users\haris\downloads\pyleecan-1.4.2.tar\pyleecan-1.4.2\pyleecan-1.4.2\pyleecan\Classes\MeshSolution.py", line 31, in <module>
    from ..Methods.Mesh.MeshSolution.get_field import get_field
  File "C:\Users\Haris\AppData\Local\Programs\Python\Python38\Lib\site-packages\shiboken2\files.dir\shibokensupport\__feature__.py", line 142, in _import
    return original_import(name, *args, **kwargs)
  File "c:\users\haris\downloads\pyleecan-1.4.2.tar\pyleecan-1.4.2\pyleecan-1.4.2\pyleecan\Methods\Mesh\MeshSolution\get_field.py", line 5, in <module>
    from ....Classes.MeshVTK import MeshVTK
  File "C:\Users\Haris\AppData\Local\Programs\Python\Python38\Lib\site-packages\shiboken2\files.dir\shibokensupport\__feature__.py", line 142, in _import
    return original_import(name, *args, **kwargs)
  File "c:\users\haris\downloads\pyleecan-1.4.2.tar\pyleecan-1.4.2\pyleecan-1.4.2\pyleecan\Classes\MeshVTK.py", line 72, in <module>
    from vtk import vtkPointSet
  File "C:\Users\Haris\AppData\Local\Programs\Python\Python38\Lib\site-packages\shiboken2\files.dir\shibokensupport\__feature__.py", line 142, in _import
    return original_import(name, *args, **kwargs)
  File "C:\Users\Haris\AppData\Local\Programs\Python\Python38\lib\site-packages\vtk.py", line 31, in <module>
    all_m = importlib.import_module('vtkmodules.all')
  File "C:\Users\Haris\AppData\Local\Programs\Python\Python38\lib\importlib\__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "C:\Users\Haris\AppData\Local\Programs\Python\Python38\lib\site-packages\vtkmodules\all.py", line 92, in <module>
    from .vtkIOCONVERGECFD import *
  File "C:\Users\Haris\AppData\Local\Programs\Python\Python38\Lib\site-packages\shiboken2\files.dir\shibokensupport\__feature__.py", line 142, in _import
    return original_import(name, *args, **kwargs)
KeyboardInterrupt
^C

BonneelP, 2023-01-06:
Hello,

Anaconda install several packages versions that may conflict with pyleecan ones. We used to recommend to use Anaconda but now we advise to use a clean python and VScode (cf https://pyleecan.org/get.pyleecan.html). In particular there were issue with Anaconda and pyleecan GUI that should still be there, so I would recommend not to use Anaconda with pyleecan if you plan to use the GUI. 

Best regards,
Pierre

crowdnidhi-design, 2023-01-06:
Sure I got rid of Anaconda and reinstalled 1.4.2. It seems to be working but I have found some other issues. Anyway I will clean it.
Where can I find detailed literature on winding, please?
I don't find details on adding the insulation paper details in the stator slot. If this property is not available what is the percentage we need to allocate in the slot fill factor?.

BonneelP, 2023-01-06:
Hello, 

Regarding winding, there are some information on the website: https://pyleecan.org/winding.convention.html
Regarding the fill factor, the slot surface is defined in step 3 "Stator Slot" then the active surface/insulation is defined at step 5 "Stator Conductor" and the fill factor is available as an output here: 
[image]

Best regards,
Pierre

crowdnidhi-design, 2023-01-24:
Dear Pierre,

Well, I'm still struggling to get the stator conductor information right.
Let me give you an example and in turn can you please show me how to fill in the details in the "Stator conductor Page" ( GUI). Please note I have entered the below in the stator winding page

Winding Pattern:
Slot Number =48
Pole pair number=4
Phases number=3
Layers number=1
Coil pitch=5
Turns per coil=60
Parallel circuits=1

I have below set of information I need to use

1. Motor has 48 slots and 8 poles ( 4 pole pairs). 3 Phase start connected motor
2.  The coil details are as below
     Number of Turns: 60
     Wire Diameter ( in mm): 0.7239
     Film thickness ( in mm): 0.0254
     Number of Strands: 2
     Insulator thickness( in mm) i.e. slot insulator paper: 0.13 
     No of wire Layers: 1

crowdnidhi-design, 2023-01-29:
Guys can some one help me?.

BonneelP, 2023-01-30:
Hello,

Sorry for the answer delay, I was in holiday last week. Regarding the winding definition you can refer to several ressources to understand pyleecan way of defining the winding:
-  https://pyleecan.org/winding.convention.html
- In the GUI all fields should have a tooltip (by hovering your cursor over the field, you should have the definition of the parameter).
- All the parameters are detailed in the corresponding csv files (https://github.com/Eomys/pyleecan/blob/master/pyleecan/Generator/ClassesRef/Machine/Winding.csv, https://github.com/Eomys/pyleecan/blob/master/pyleecan/Generator/ClassesRef/Machine/Conductor.csv, https://github.com/Eomys/pyleecan/blob/master/pyleecan/Generator/ClassesRef/Machine/CondType12.csv)
- The code of the GUI detail which method is used for each output so you can check how it is computed: https://github.com/Eomys/pyleecan/blob/master/pyleecan/GUI/Dialog/DMachineSetup/SWindCond/WCondOut/WCondOut.py

Which parameter are you having issue with ?

Best regards,
Pierre

crowdnidhi-design, 2023-02-07:
Dear Pierre,

I hope you had a great vacation/holiday.
Well I 'am trying to fill in the below details in Pyleecan GUI
1. Stator winding - Step 4 in GUI

Please note I have entered the below in the stator winding page

Winding Pattern:
Slot Number =48
Pole pair number=4 ( 8 poles)
Phases number=3
Layers number=1
Coil pitch=5
Turns per coil=60
Parallel circuits=1

This is absolutely clear to me in Pyleecan.

2. Stator Conductor - Step 5 in GUI

I am using the below link to enter magnet wire details 

https://www.elektrisola.com/en-us/Products/Enamelled-Wire/Technical-Data#iec60317
EUROPE / ASIA IEC 60317

For example, if I want to use 2 strands of wire, 0.4 mm ( nominal diameter). Can you please let me know what should we fill in the fields displayed on Step 5 GUI ( Stator Conductor)?

Can you post a picture after you enter the details?

Regards,
Harish

BonneelP, 2023-02-09:
Hello,

Based on your description, I guess that the correct parameters are Nwppc = 2 and Wwire = 0.4. Insulation parameters should be optional to set. 
To check the definition you can use the output below. The important data to check is the Ksfill and the active surface (Scond_active). Once both these output are correct, you can then set Lewout (endwinding length) to correct the resistance and winding mass. 

Best regards,
Pierre

crowdnidhi-design, 2023-05-11:
I will review it once thanks