Eomys/pyleecan #591: The simulation is terminating abruptly with below error ...

crowdnidhi-design, 2023-01-06:
[19:17:09] Error while saving results:
module 'numpy' has no attribute 'warnings'
[19:17:13] Error while saving results:
module 'numpy' has no attribute 'warnings'
[19:17:15] Error while plotting flux:
[Errno 2] No such file or directory: 'C:/Users/Haris/Downloads/2023_01_06-19h16min51s_FEMM_POC9-01062023-v1_radial.6/flux as fct of time.png'

Iam not sure why this error is creeping up please..

BonneelP, 2023-01-06:
Hello,
I don't have this issue on my side and sadly I don't have much time to investigate. I would expect a package version mismatch on your side. Maybe you can try to run "pip install -U -r requirements.txt" in the main folder ? There is also the GUI exe on the website that should work. 
Regarding the last message maybe the "." in the result name in the reason.
Best regards,
Pierre

crowdnidhi-design, 2023-01-06:
yes ur right "." if we use period in saving files it is creating issues..
"module 'numpy' has no attribute 'warnings'" is manageable I will fix this.

crowdnidhi-design, 2023-01-06:
C:\Users\Haris>pip install --force-reinstall numpy==1.23.5
Collecting numpy==1.23.5
  Downloading numpy-1.23.5-cp38-cp38-win_amd64.whl (14.7 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 14.7/14.7 MB 11.3 MB/s eta 0:00:00
Installing collected packages: numpy
  Attempting uninstall: numpy
    Found existing installation: numpy 1.21.4
    Uninstalling numpy-1.21.4:
      Successfully uninstalled numpy-1.21.4
Successfully installed numpy-1.23.5

C:\Users\Haris>python -m pyleecan
[21:04:07] Saving Simu1 to file 'C:/Users/Haris/Downloads/2023_01_06-21h04min07s_FEMM_POC9-01062023-v1_6\FEMM_POC9-01062023-v1_6.json'.
Saving Simu1 to file 'C:/Users/Haris/Downloads/2023_01_06-21h04min07s_FEMM_POC9-01062023-v1_6\FEMM_POC9-01062023-v1_6.json'.
[21:04:07] Starting running simulation FEMM_POC9-01062023-v1_6 (machine=POC9-01062023-v1_6)
[21:04:07] Starting Magnetic module
[21:04:08] Computing Airgap Flux in FEMM
[21:04:19] End of simulation FEMM_POC9-01062023-v1_6

Issue fixed .. seems to be working now

closed 2023-01-16 by BonneelP