Eomys/pyleecan #592: ImportError: cannot import name 'FEMMHandler'

crowdnidhi-design, 2023-01-08:
ImportError                               Traceback (most recent call last)
Cell In[1], line 22
     19 example1.plot()
     21 # Check FEMM installation
---> 22 from pyleecan.Classes._FEMMHandler import FEMMHandler
     24 femm = FEMMHandler()
     25 femm.openfemm(0)

ImportError: cannot import name 'FEMMHandler' from 'pyleecan.Classes._FEMMHandler' (c:\users\haris\downloads\pyleecan-1.4.2.tar\pyleecan-1.4.2\pyleecan-1.4.2\pyleecan\Classes\_FEMMHandler.py)

crowdnidhi-design, 2023-01-08:
how to solve the above issue pls

BonneelP, 2023-01-16:
Hello, 

Do you still have this error ? I think it may be just a typo "FEMMHandler" instead of "_FEMMHandler"

Best regards,
Pierre

crowdnidhi-design, 2023-02-07:
No. I have resolved it.

closed 2023-04-28 by BonneelP