Eomys/pyleecan #593: ---> 31 varload.set_OP_matrix(OP_matrix, "N0", "I0", "Phi0", "Tem") - I am getting an error . Can someone help

crowdnidhi-design, 2023-01-09:
AttributeError                            Traceback (most recent call last)
Cell In[24], line 31
     29 # Set reference torque from Yang et al, 2013
     30 OP_matrix[:,3] = Tem_av_ref
---> 31 varload.set_OP_matrix(OP_matrix, "N0", "I0", "Phi0", "Tem")
     32 print(OP_matrix)
     34 # Add DataKeeper to track I0/Phi0/Tem

AttributeError: 'VarLoadCurrent' object has no attribute 'set_OP_matrix'

SebGue, 2023-01-09:
Hello,

I guess you are trying to run some of the tutorials. Unfortunately they seem to be outdated, i.e. there is no VarLoadCurrent.set_OP_matrix method any more.
Now, to set the operation points you could adapt the following code

```
Nspeed = 2
OP_matrix = np.zeros((Nspeed, 5))
OP_matrix[:, 0] = np.array([3000, 10000]) # speed
OP_matrix[:, 4] = np.array([30*1e3, 36*1e3]) # power
simu.var_simu = VarLoadCurrent(datakeeper_list=datakeeper_list)
simu.var_simu.set_OP_array(OP_matrix, "N0", "Id", "Iq", "Tem", "Pem")
```

Best regards, Sebastian

crowdnidhi-design, 2023-01-09:
Yes, Sebastian ur right. Can I know where are the updated tutorials? and where is the updated documentation for Classes, functions and methods, etc

crowdnidhi-design, 2023-01-09:
Iam getting below error
NameError: name 'np' is not defined

crowdnidhi-design, 2023-01-09:
solved np issue but iam getting different errors pls.
Can you please list all the classes required.

SebGue, 2023-01-10:
For my example code you need
`from pyleecan.Classes.VarLoadCurrent import VarLoadCurrent` and
`import numpy as np`
Further you need a datakeeper_list, but you can also skip the datakeepers depending on your code.

crowdnidhi-design, 2023-01-10:
# Dear Sebatian. I have tried to incorporate your code. The issue is some time it works and some time I get error "  NameError: name 'simu' is not defined". Iam not sure what is wrong. Iam using "IPMSM_B.json"

from pyleecan.Classes.VarLoadCurrent import VarLoadCurrent
from pyleecan.Classes.DataKeeper import DataKeeper
from pyleecan.Classes.VarParamSweep import VarParamSweep
from pyleecan.Classes.VarSimu import VarSimu
from pyleecan.Classes.Simu1 import Simu1
from pyleecan.Classes.ParamExplorerInterval import ParamExplorerInterval
from numpy import zeros, ones, linspace, array, sqrt, arange
import numpy as np

datakeeper_list = []

simu_vop = simu_ref.copy()
simu_vop.mag.import_file = None
varload = VarLoadCurrent()
simu_vop.var_simu = varload

# Creating the Operating point matrix
OP_matrix = zeros((N_speed,4))

# Set N0 = 2000 [rpm] for all simulation
OP_matrix[:,0] = 2000 * ones((N_speed))

# Set I0 = 250 / sqrt(2) [A] (RMS) for all simulation
OP_matrix[:,1] = 250/sqrt(2) * ones((N_speed)) 

# Set Phi0 from 60° to 180°
OP_matrix[:,2] = Phi0_ref

# Set reference torque from Yang et al, 2013
OP_matrix[:,3] = Tem_av_ref

print("Operating Point matrix:")
print(OP_matrix)
print("Original Operating point:")
print(simu_vop.input.OP.get_I0_Phi0())

# Column name are given in the order of OP_matrix
#varload.set_OP_matrix(OP_matrix, "N0", "I0", "Phi0", "Tem")

Nspeed = 2
OP_matrix = np.zeros((Nspeed, 5))
OP_matrix[:, 0] = np.array([3000, 10000]) # speed
OP_matrix[:, 4] = np.array([30*1e3, 36*1e3]) # power
simu.var_simu = VarLoadCurrent(datakeeper_list=datakeeper_list)
simu.var_simu.set_OP_array(OP_matrix, "N0", "Id", "Iq", "Tem", "Pem")

# The operating point of the simulation is automatically set as
# the first one from the OP_matrix
print("Updated Operating point:")
print(simu_vop.input.OP.get_I0_Phi0())

# All the simulation use the same machine
# No need to draw the machine for all OP
varload.is_reuse_femm_file=True

SebGue, 2023-01-10:
Is the above your simulation script? (By the way, you can highlight code with CTRL+E shortcut or upload your whole file for better readability.)
There is some mismatch in OP_matrix size between your code and my example. Further there are some variables, i.e. Tem_ref, Phi0_ref and simu_ref that seems not to be defined.

crowdnidhi-design, 2023-01-10:
This is a WIP but the code I have shared is available in this file.
I am working on a master file that has end-to-end features/functionality testing/simulations capabilities

SebGue, 2023-01-10:
Hello,

This is a lot of code and I don't exactly know where your errors appear, since I haven't tested it. 
But as I told, you should adapt my example code to yours, i.e. for your code something like
`varload.set_OP_array(OP_matrix, "N0", "I0", "Phi0", "Tem")` may work.
If you still encounter issues some minimal examples would be great :-)

Best regards, Sebastian