Eomys/pyleecan #594: Webinar n°2: Pyleecan Advanced - Part 1: Optimization

crowdnidhi-design, 2023-01-09:
Dear All,

Do you have an updated version of Webinar n°2: Pyleecan Advanced - Part 1: Optimization
I want to understand more about the simulation capabilities.
Can you please share the updated document?..

Regards,
Harish

BonneelP, 2023-01-09:
Hello,

We haven't done a second edition of the webinar and we haven't maintained the notebook as they were mostly duplicate of existing tutorials. So there is no "updated document" regarding these webinars.
For the state of the package, you can have a look at the readme file or the home page of the website that were updated in September 2022. For feature added since, you can refer to the release note. 

Best regards,
Pierre

crowdnidhi-design, 2023-01-09:
Sure, Pierre. You are right I will have to spend some time to understand the updates and work on using them.
I have got the majority of the features/functionality developed( in a notebook) except simulations.
Further, I will keep you posted if I need any help with linking Pyleecan with Scilab/Xcos.

Regards,
Harish

BonneelP, 2023-01-16:
Hello,

Yes don't hesitate to open a new issue if you have further questions.

Best regards,
Pierre

closed 2023-01-16 by BonneelP