Eomys/pyleecan #595: Id Iq simultaions

crowdnidhi-design, 2023-01-10:
[image]

crowdnidhi-design, 2023-01-10:
Dear Pierre,

Can we use the above eqs for the Id and Iq current generation?

crowdnidhi-design, 2023-01-11:
Dear All,

Any feedback on the above?

BonneelP, 2023-01-11:
Hello,

The normal process is to directly set input.OP.Id_ref and input.OP.Iq_ref then pyleecan will compute Is automatically. Otherwise you can set input.Is_enforced to set your currents but then it may require to check if everything is ok regarding time and rotor position (it should be). The corresponding methods are :
- pyleecan\Methods\Simulation\InputCurrent\gen_input.py
- pyleecan\Methods\Output\OutElec\get_Is.py

Best regards,
Pierre