Eomys/pyleecan #596: Tutorial - https://www.youtube.com/watch?v=Kj-DJGpbBkU&t=1185s

crowdnidhi-design, 2023-01-11:
Dear Pierre,

I was watching your tutorial to build simulations.
I understand that in your tutorial you have to built a strategy to change the pole angle ( W4) and magnet length( W3).
You are able to get the values and modify them as these variables are predefined in hole type 53. 
But when we develop our own design of stator and rotor and import( via DXF format) it into pyleecan how can we read all the variables that pyleecan assigns for magnets, holes and angle parameters.
I believe pyleecan assigns some numbers to arcs and segments ...
Can you please educate me on this. 

Regards,
Harish

crowdnidhi-design, 2023-01-12:
Dear Pierre,

Are you there please? 
Do you have any updates for me?.....

BonneelP, 2023-01-12:
Hello,

Sorry I don't have much time to work on pyleecan right now so I take longer to answer issues.

In pyleecan, the best way to know what a class contains is to look to the csv files that contains the description of all pyleecan classes (pyleecan/Generator/ClassesRef) . In particular all user defined classes have "UD" in the name. You can take a look at:
- SlotUD (active surface closed by a line - GUI version)
- SlotUD2 (active surface defined by surface - not available in the GUI) 
- HoleUD (for IPMSM rotor). 

You will find "surf_list" and "line_list" that store "SurfLine" objects with "Segment" and "Arc" that describes your geometry.

Hope it helps,
Best regards,
Pierre

closed 2023-01-16 by BonneelP