Eomys/pyleecan #597: Request for information

crowdnidhi-design, 2023-01-24:
Dear All,

Can I pull the below information?. 
Do we have readily available variables/functions/methods?

Power, in Watts
Power factor
Efficiency
Wire Current Density, A/m2
Copper Loss in Watts
Iron Loss in Watts
Coil Current Density A/mm2
Line Voltage(peak), V

BonneelP, 2023-01-30:
Hello,

As a first short answer without much investigation:
- "Average Electromagnetic power [W]" output.mag.comp_power() : https://github.com/Eomys/pyleecan/blob/master/pyleecan/Methods/Output/OutMag/comp_power.py
- For losses, it requires to define additional model that will be extended/detailed in PR #558
- Line voltage shouldn't be available since most of the simulation are current driven. We supply directly the current as a fct of time in the FEMM models, in the FEMM related methods/functions you should find the current density.

Hope it helps, 
Best regards,
Pierre

crowdnidhi-design, 2023-02-07:
Thanks Pierre.

closed 2023-04-28 by BonneelP