Eomys/pyleecan #598: Problem with tutorial "Iterating on several Operating Point"

ndsharma1, 2023-01-26:
Hello

"Pyleecan" is a wonderful product, and is making my life a lot easier when I try to simulate SPMS motors. Thanks a lot for the initiative.
When I try to run the "Iterating on several Operating Point" from the tutorial "https://www.pyleecan.org/tuto_Operating_point.html" , I get following error. 

I tried checking on the attribute list of the 'ValLoadCurrent' class... could not find the needed attribute. Please help me move past this error.

---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
~\AppData\Local\Temp\ipykernel_11892\1967870894.py in <module>
     28 print(simu_vop.input.OP.get_I0_Phi0())
     29 # Column name are given in the order of OP_matrix
---> 30 varload.set_OP_matrix(OP_matrix, "N0", "I0", "Phi0", "Tem")
     31 # The operating point of the simulation is automatically set as
     32 # the first one from the OP_matrix (to speed-up computation due to how pyleecan works)

AttributeError: 'VarLoadCurrent' object has no attribute 'set_OP_matrix'

ndsharma1, 2023-01-26:
import numpy as np
Nspeed = 2
OP_matrix = np.zeros((Nspeed, 5))
OP_matrix[:, 0] = np.array([3000, 10000]) # speed
OP_matrix[:, 4] = np.array([30*1e3, 36*1e3]) # power
#simu.var_simu = VarLoadCurrent(datakeeper_list=datakeeper_list)
simu_op.var_simu.set_OP_array(OP_matrix, "N0", "Id", "Iq", "Tem", "Pem")
import numpy as np
Nspeed = 2
Just saw a proposed solution in another thread. This also resulted in similar error: 

**I tried this method:**

OP_matrix = np.zeros((Nspeed, 5))
OP_matrix[:, 0] = np.array([3000, 10000]) # speed
OP_matrix[:, 4] = np.array([30*1e3, 36*1e3]) # power
#simu.var_simu = VarLoadCurrent(datakeeper_list=datakeeper_list)
simu_op.var_simu.set_OP_array(OP_matrix, "N0", "Id", "Iq", "Tem", "Pem")

**Resulted in following error**
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
~\AppData\Local\Temp\ipykernel_11892\149860735.py in <module>
      5 OP_matrix[:, 4] = np.array([30*1e3, 36*1e3]) # power
      6 #simu.var_simu = VarLoadCurrent(datakeeper_list=datakeeper_list)
----> 7 simu_op.var_simu.set_OP_array(OP_matrix, "N0", "Id", "Iq", "Tem", "Pem")

AttributeError: 'NoneType' object has no attribute 'set_OP_array'

ndsharma1, 2023-01-26:
I just saw a proposed solution in another thread. I tried that version, but still getting error. Please see below. 
Kindly help.

from numpy import linspace, array, pi
Tem_av_ref = array([0,12.5, 24.7, 35, 40.5, 45]) # Yang et al, 2013
Phi0_ref = linspace(60 * pi / 180, 180 * pi / 180, Tem_av_ref.size)
N_simu = Tem_av_ref.size

_# Create new Variable speed simulation_
simu_vop = simu_op.copy()
simu_vop.mag.import_file = None
varload = VarLoadCurrent()
simu_vop.var_simu = varload

_# Creating the Operating point matrix_
OP_matrix = zeros((N_simu,5))

_# Set N0 = 450 [rpm] for all simulation_
OP_matrix[:,0] = [0,150,250,350,450,500]

_# Set I0 = 10 / sqrt(2) [A] (RMS) for all simulation_
OP_matrix[:,1] = I0_rms * ones((N_simu))

_# Set Phi0 from 60° to 180°_
OP_matrix[:,2] = Phi0_ref

_# Set reference torque from Yang et al, 2013_
OP_matrix[:,3] = Tem_av_ref
OP_matrix[:, 4] = [0, 196.35, 646.6, 1282.8, 1908.5, 2356.2] # power

print("Operating Point matrix:")
print(OP_matrix)
print("Original Operating point:")
print(simu_vop.input.OP.get_I0_Phi0())
_# Column name are given in the order of OP_matrix_
varload.set_OP_array(OP_matrix, "N0", "I0", "Phi0", "Tem")
_# The operating point of the simulation is automatically set as_
_# the first one from the OP_matrix (to speed-up computation due to how pyleecan works)_
print("Updated Operating point:")
print(simu_vop.input.OP.get_I0_Phi0())

_# All the simulation use the same machine_
_# No need to draw the machine for all OP_
varload.is_reuse_femm_file=True

_# Add I0/Phi0 to the data available at the end of the simulation_
I0_dk = DataKeeper(
    name="Stator current rms amplitude",
    symbol="I0",
    unit="Arms",
    keeper="lambda output: output.elec.OP.get_I0_Phi0()['I0']",
)
Phi0_dk = DataKeeper(
    name="Stator current phase",
    symbol="Phi0",
    unit="rad",
    keeper="lambda output: output.elec.OP.get_I0_Phi0()['Phi0']",
)
varload.datakeeper_list= [I0_dk, Phi0_dk]

Xout = simu_vop.run()

**I get following error:**

[12:35:51] Starting running simulation FEMM_XXXXXXX
---------------------------------------------------------------------------
TypeError                                 Traceback (most recent call last)
~\AppData\Local\Temp\ipykernel_11892\2456258834.py in <module>
----> 1 Xout = simu_vop.run()

C:\ProgramData\Anaconda3\lib\site-packages\pyleecan\Methods\Simulation\Simulation\run.py in run(self)
     36 
     37         # Compute the multisimulation
---> 38         self.var_simu.run()
     39 
     40     # 'normal' simulation

C:\ProgramData\Anaconda3\lib\site-packages\pyleecan\Methods\Simulation\VarSimu\run.py in run(self)
     37 
     38     # Generate simulation list and ParamExplorerValue list
---> 39     simu_dict = self.generate_simulation_list(ref_simu)
     40     # Generate default datakeeper
     41     self.gen_datakeeper_list(ref_simu)

C:\ProgramData\Anaconda3\lib\site-packages\pyleecan\Methods\Simulation\VarLoadCurrent\generate_simulation_list.py in generate_simulation_list(self, ref_simu)
     30     #   Other parameters can be stored in a dedicated ParamExplorerSet if needed
     31     multisim_dict["paramexplorer_list"].append(
---> 32         ParamExplorerSet(
     33             name="InputCurrent",
     34             symbol="In",

C:\ProgramData\Anaconda3\lib\site-packages\pyleecan\Classes\ParamExplorerSet.py in __init__(self, value, name, symbol, unit, setter, getter, init_dict, init_str)
    143         self.value = value
    144         # Call ParamExplorer init
--> 145         super(ParamExplorerSet, self).__init__(
    146             name=name, symbol=symbol, unit=unit, setter=setter, getter=getter
    147         )

C:\ProgramData\Anaconda3\lib\site-packages\pyleecan\Classes\ParamExplorer.py in __init__(self, name, symbol, unit, setter, getter, init_dict, init_str)
    125         self.symbol = symbol
    126         self.unit = unit
--> 127         self.setter = setter
    128         self.getter = getter
    129 

C:\ProgramData\Anaconda3\lib\site-packages\pyleecan\Classes\_frozen.py in __setattr__(self, key, value)
     33                 self.__class__.__name__ + ' class has no "' + key + '" ' "property"
     34             )
---> 35         object.__setattr__(self, key, value)
     36 
     37     def _freeze(self):

TypeError: 'property' object is not callable

BonneelP, 2023-01-30:
Hello,

Sorry for the delay to answer, I was in holiday last week.
Your error message is rather strange. It looks like an internal error on a part that have a fair amount of tests. I will try to investigate when I will have some time. Which version of pyleecan/python are you using ?

Best regards,
Pierre

ndsharma1, 2023-01-30:
Pierre
Thanks for your response.
I am using 
Python version: 3.9.13 (Jupyter Notebook)
Pyleecan version: 1.4.2
SciDataTool version: 2.5.0

BonneelP, 2023-04-28:
Hello,

I just release pyleecan1.5.0 and updated all the tutorials on the website (there is a new loss tutorial). Normally the DataKeeper issue should be solved. 
Do you need further help or can we close this issue ? 

Best regards,
Pierre