Eomys/pyleecan #599: Changes in rotor speed (N0) don't affect  average Torque

woland51, 2023-02-13:
Hello, 
I'm currently working on a University project for my Electrical Machines course and I have to model some typologies of IPMSM using Pyleecan.
To overcome Datakeeper's issues I've wrote down a simple for loop iteranting over a bunch of lists I provide as inputs (current, angle, speed), unfortunately providing different rotor speeds  doesn't affect average torque (I even tried to reach  10000 rpm and no changes in torque).
Is there a way to link rotor speed to average torque without using Datakeeper structure? 
Is it possible to plot  the motor mechanical characteristic (x: angular rotor speed, y: mechanical torque)?

```
rotor_speed = 500 # [rpm] 
rotor_speed_list = [number + rotor_speed for number in range(0,1000,83)]
Id_0k = [number + 200 for number in range(0,260,20)]  # input current list
Phi_0k= [number  for number in range(60,125,5)]
Tem_av = []

for i in range(len(Id_0k)):

    simulation.input = InputCurrent()  # Definition of a sinusoidal current
    I0_rms = Id_0k[0]/sqrt(2) # Maximum current [Arms]  # I0, Phi0 to set
    Phi0 = Phi_0k[0]*pi/180  # Maximum Torque Per Amp
    Id_ref = (I0_rms*exp(1j*Phi0)).real # [Arms] # Compute corresponding Id/Iq
    Iq_ref = (I0_rms*exp(1j*Phi0)).imag # [Arms]
    speed =  rotor_speed_list[i]
    # Setting the values
    simulation.input.OP = OPdq(Id_ref=Id_ref, Iq_ref=Iq_ref, N0 = speed)  # OP definition
    (Id_ref,Iq_ref)
    simulation.input.Nt_tot = 128 # Number of time step
    simulation.input.Na_tot = 2048 # Spatial discretization
    out_op = simulation.run()
    Tem_av_i = out_op.mag.Tem_av  
    Tem_av.append(Tem_av_i)

dict_data_set = {"speed": rotor_speed_list, "Id_0" : Id_0k[0], "Phi_0": Phi_0k[0], "Average_Tem": Tem_av}
```

<html xmlns:o="urn:schemas-microsoft-com:office:office"
xmlns:x="urn:schemas-microsoft-com:office:excel"
xmlns="http://www.w3.org/TR/REC-html40">

<head>

<meta name=ProgId content=Excel.Sheet>
<meta name=Generator content="Microsoft Excel 15">
<link id=Main-File rel=Main-File
href="file:///C:/Users/Lenovo/AppData/Local/Temp/msohtmlclip1/01/clip.htm">
<link rel=File-List
href="file:///C:/Users/Lenovo/AppData/Local/Temp/msohtmlclip1/01/clip_filelist.xml">
<style>

</style>
</head>

<body link=blue vlink=purple>

speed | Id_0 | Phi_0 | Average_Tem
-- | -- | -- | --
500 | 200 | 60 | 14,58983
583 | 200 | 60 | 14,58983
666 | 200 | 60 | 14,58983
749 | 200 | 60 | 14,58983
832 | 200 | 60 | 14,58983
915 | 200 | 60 | 14,58983
998 | 200 | 60 | 14,58983
1081 | 200 | 60 | 14,58983
1164 | 200 | 60 | 14,58983
1247 | 200 | 60 | 14,58983
1330 | 200 | 60 | 14,58983
1413 | 200 | 60 | 14,58983
1496 | 200 | 60 | 14,58983

</body>

</html>

BonneelP, 2023-02-13:
Hello,

Great to know that some university are using Pyleecan for courses :) Could you share the name of your university and professor ? We are looking for feedback/collaboration on how to improve pyleecan for university. 

What is your issue with Datakeeper ? Is there an error message ? Which version of pyleecan are you using ?

The loop you have construct is the purpose of the object VarLoadCurrent that would speed up your calculation (VarLoadCurrent should draw your machine in FEMM only once while you draw it at each call to simu.run). You can take a look at https://pyleecan.org/tuto_Operating_point.html#Iterating-on-several-Operating-Point.

As you are using InputCurrent, the simulation is current driven for an IPMSM then the torque doesn't change with speed (there is no electrical model in this case).

Best regards,
Pierre

woland51, 2023-02-13:
Hi Pierre, 
thanks for your reply :)
I'm studying at UNIVPM ( Università Politecnica delle Marche, Ancona, Italy) with Professor [Emanuele Principi](https://www.univpm.it/Entra/Ingegneria_1/docname/idsel/820/docname/EMANUELE%20PRINCIPI) 
So, basically, simulations are only current or voltage driven? 
Since I have to compute rotor speed for every average torque value at a certain input current, introducing an efficiency factor in the OPdq constructor and than computing speed using Power = (Torque*angular speed) will work just fine?
Referring to Datakeeper I had the same problem as  issue #598.
I'm running:
pyleecan                 1.4.2
pyfemm                   0.1.3
SciDataTool              2.5.0
Python                     3.9.7
Thanks for your help

BonneelP, 2023-04-28:
Hello,

I just release pyleecan1.5.0 and updated all the tutorials on the website (there is a new loss tutorial). Normally the DataKeeper issue should be solved. 
Do you need further help or can we close this issue ? 

Best regards,
Pierre

woland51, 2023-05-02:
Hello Pierre, 
nice to hear these news of yours, I'll take a look at the new features as soon as I can.
Of course you can close the issue.
Best regards, 
Giuliano

closed 2023-05-02 by BonneelP