Eomys/pyleecan #617: post installation tests

Elektro7, 2023-04-25:
hello im trying to get pyleecan working. im using python 3.8.10 also i cloned pyleecan repo. and i downloaded all requirements through pip.

im assuming if i run the command : python -m pytest ./Tests -m "not long_5s".
then i should be getting 0 failed tests. however im getting 34 failed tests.

this is the summary

=========================== short test summary info ===========================
FAILED Tests/Classes/test_classes.py::test_class_methods[class_dict103] - Imp...
FAILED Tests/Classes/test_classes.py::test_class_methods[class_dict104] - Imp...
FAILED Tests/Classes/test_classes.py::test_class_methods[class_dict105] - Imp...
FAILED Tests/Classes/test_classes.py::test_class_methods[class_dict135] - Imp...
FAILED Tests/Classes/test_classes.py::test_class_methods[class_dict136] - Imp...
FAILED Tests/Classes/test_classes.py::test_class_methods[class_dict137] - Imp...
FAILED Tests/Classes/test_classes.py::test_class_methods[class_dict149] - Imp...
FAILED Tests/Classes/test_classes.py::test_class_methods[class_dict151] - Imp...
FAILED Tests/Functions/test_save_load.py::test_save_load_simu[h5] - Attribute...
FAILED Tests/Functions/test_save_load.py::test_save_load_slash_h5 - Attribute...
FAILED Tests/GUI/Dialog/DMatLib/test_DMatSetup.py::TestDMatSetup::test_BH_setup
FAILED Tests/Methods/Import/test_import_mesh_unv.py::test_import_mesh_unv[unv_file0]
FAILED Tests/Methods/Import/test_import_mesh_unv.py::test_import_mesh_unv[unv_file1]
FAILED Tests/Methods/Import/test_import_mesh_unv.py::test_import_mesh_unv[unv_file2]
FAILED Tests/Methods/Import/test_import_mesh_unv.py::test_import_mesh_unv[unv_file3]
FAILED Tests/Methods/Import/test_import_mesh_unv.py::test_import_mesh_unv[unv_file4]
FAILED Tests/Methods/Mesh/test_perm_coord.py::test_perm_coord - ImportError: ...
FAILED Tests/Methods/Mesh/test_plot_contour.py::test_plot_contour_1group - Im...
FAILED Tests/Methods/Mesh/test_plot_contour.py::test_plot_contour_2group - Im...
FAILED Tests/Methods/Mesh/test_plot_mesh.py::test_plot_mesh - ImportError: ca...
FAILED Tests/Methods/Mesh/test_plot_meshsol.py::test_plot_glyph - ImportError...
FAILED Tests/Methods/Mesh/test_plot_meshsol.py::test_plot_deflection - Import...
FAILED Tests/Methods/Mesh/Mesh/test_convert_MeshVTK.py::test_convert_MeshVTK
FAILED Tests/Methods/Simulation/test_VarParamSweep_fct_save_load.py::test_VarParamSweep_fct_save_load
= 24 failed, 5400 passed, 16 skipped, 93 deselected, 120 warnings in 358.84s (0:05:58) =

im also having an issue through the tutorial "how to define a simulation to call femm".

when i call this command :
out_femm.mag.meshsolution.plot_contour(label="B", group_names="stator core") i get an error relating to vtk. i guess the error is related to the following issue.
https://github.com/pyvista/pyvista/issues/2580

im looking forward for your feedback!, thank you in advance :)

Elektro7, 2023-04-25:
so i could reduced the errors to 6 errors through reinstalling pyvista. however its version 0.38.5. so i get this 

ERROR: pip's dependency resolver does not currently take into account all the packages that are installed. This behaviour is the source of the following dependency conflicts.
pyleecan 1.4.2 requires pyvista<=0.31.3,>=0.25.3, but you have pyvista 0.38.5 which is incompatible.

after running python -m pytest ./Tests -m "not long_5s". i get now these failed tests.

FAILED Tests/Functions/test_save_load.py::test_save_load_simu[h5] - Attribute...
FAILED Tests/Functions/test_save_load.py::test_save_load_slash_h5 - Attribute...
FAILED Tests/GUI/Dialog/DMatLib/test_DMatSetup.py::TestDMatSetup::test_BH_setup
FAILED Tests/Methods/Mesh/test_plot_meshsol.py::test_plot_glyph - TypeError: ...
FAILED Tests/Methods/Mesh/test_plot_meshsol.py::test_plot_deflection - pyvist...
FAILED Tests/Methods/Simulation/test_VarParamSweep_fct_save_load.py::test_VarParamSweep_fct_save_load
= 6 failed, 5418 passed, 16 skipped, 93 deselected, 119 warnings in 356.92s (0:05:56) =

BonneelP, 2023-04-25:
Hello,

Thank you for your interest in pyleecan. I just checked and tests are running fine on my side on the master branch. I'm using python 3.8.10 on a windows 11 with pyvista 0.31.3 can you check with this version ?

Can you detail the error messages you are receiving ? We should soon release the next version of pyleecan, if you find something to update in the requirements we can add the correction to it.

Best regards,
Pierre

Elektro7, 2023-04-25:
thank you for your reply.  im also on windows 11, python 3.8.10 and now i installed pyvista 0.38.1 and openpyxl which resolved alot of issues. however i see there is a common error which is the following:

>       raise AttributeError("module {!r} has no attribute "
                             "{!r}".format(__name__, attr))
E       AttributeError: module 'numpy' has no attribute 'warnings'

im a beginner in python, so im not exactly sure what this error mean.

Nasser

Elektro7, 2023-04-26:
I was able to reduce the errors to zero, I just downgraded Vtk and numpy to  previous versions. However I’m still getting this warning while running „calling Femm simulation“.

out_femm.mag.meshsolution.plot_contour(label="B", group_names="stator core")
gives this warning

WARNING:root:VTK requires 3D points, but 2D points given. Appending 0 third component.
however the plot opens but the results are wrong.

I’m not sure if I should open a new issue for this ? 

Thanks
Nasser

BonneelP, 2023-04-26:
Hello,

I think that the error message is "normal". We always have some struggle with the vtk plots and the jupyter notebook. Can you share the wrong plot result ? Did you change something in the notebook ?

Best regards,
Pierre

Elektro7, 2023-04-26:
hello,

im actually following the tutorial. i can see that the scaling is also different from the one expected
[image]

BonneelP, 2023-04-27:
Hello,

I'm updating the tutorials and here are some observations/explanations:
- The image is on pyleecan website so it is rather old
- The image has more element than the default mesh, so it is not the result of that jupyter notebook plot 
- The colormap on your screenshot is from -2.48 to 2.8. I have the same on my side, when setting colormap from 0 to 3, I get something closer to the website image.
- If I remember well at some point we changed the Prius winding.

[image]

Best regards,
Pierre

BonneelP, 2023-04-28:
Hello,

I just release pyleecan1.5.0 and updated all the tutorials on the website (there is a new loss tutorial).
Do you need further help or can we close this issue ? 

Best regards,
Pierre