Eomys/pyleecan #620: optimization

faraz4714, 2023-05-10:
hello
I have this error 
can you help me?

Best regards,
faraz mirali

```
from pyleecan.Classes.OptiGenAlgNsga2Deap import OptiGenAlgNsga2Deap

# Solve problem with NSGA-II
solver = OptiGenAlgNsga2Deap(problem=my_prob, size_pop=16, nb_gen=8, p_mutate=0.5)
res = solver.solve()
```

---------------------------------------------------------------------------
```
ImportError                               Traceback (most recent call last)
Cell In[7], line 5
      3 # Solve problem with NSGA-II
      4 solver = OptiGenAlgNsga2Deap(problem=my_prob, size_pop=16, nb_gen=8, p_mutate=0.5)
----> 5 res = solver.solve()

File c:\users\faraz\appdata\local\programs\python\python38\lib\site-packages\pyleecan\Classes\OptiGenAlgNsga2Deap.py:83, in OptiGenAlgNsga2Deap.<lambda>(x)
     79 # Check ImportError to remove unnecessary dependencies in unused method
     80 # cf Methods.Optimization.OptiGenAlgNsga2Deap.solve
     81 if isinstance(solve, ImportError):
     82     solve = property(
---> 83         fget=lambda x: raise_(
     84             ImportError("Can't use OptiGenAlgNsga2Deap method solve: " + str(solve))
     85         )
     86     )
     87 else:
     88     solve = solve

File c:\users\faraz\appdata\local\programs\python\python38\lib\site-packages\pyleecan\Classes\_check.py:347, in raise_(ex)
    345 def raise_(ex):
    346     """Function to raise an exeption for the method import lambda"""
--> 347     raise ex

ImportError: Can't use OptiGenAlgNsga2Deap method solve: No module named 'deap'
```

BonneelP, 2023-05-10:
Hello,

deap is an optional requirements for pyleecan, required only if you are using the optimization with NSGA2 (there is also the Bayes one that uses smoot instead).
You can solve that with a pip install deap. 

Best regards,
Pierre

faraz4714, 2023-06-19:
Hello Pierre, 
thank you for your help in advance,
I got confused and I don't know how you define evaluate_fct or Tem_av or Tem_rip_norm(what is a formula that you used?) and what is the input for our simulation for example current, rotor speed, magnet temperature, or .... and how we can optimize the parameters of the motor for example tooth of slot of stator  
can you explain more about this part of your code, please?
```
`%matplotlib inline
import numpy as np
from pyleecan.Classes.Simu1 import Simu1
from pyleecan.Classes.InputCurrent import InputCurrent

# Create the Simulation without machine
simu_ref = Simu1(name="Tuto_Opti_Bayes", machine=None)   

# Defining Simulation Input
Ir= np.zeros(30)  # Ir will be the design variables
simu_ref.input = InputCurrent(Ir=Ir)

# No models
simu_ref.elec = None
simu_ref.mag = None
simu_ref.force = None
simu_ref.struct = None 

# Evaluation function for Zitzler–Deb–Thiele's function N.3
# Using Ir as input, Tem_av and Tem_rip_norm as output
def evaluate_fct(output):
    x = output.simu.input.Ir.value
    f1 = lambda x: x[0]
    g = lambda x: 1 + (9 / 29) * np.sum(x[1:])
    h = lambda f1, g: 1 - np.sqrt(f1 / g) - (f1 / g) * np.sin(10 * np.pi * f1)
    output.mag.Tem_av = f1(x)
    output.mag.Tem_rip_norm = g(x) * h(f1(x), g(x))`
```

best regards,
Faraz Mirali

BonneelP, 2023-06-19:
Hello,

This test is a proof of concept to check the algorithm of the optimization without pyleecan. evaluate_function enables to not call pyleecan usual function. In this very odd exemple, in fact Tem_av and Tem_rip_norm are used to stored something completely different (for the sake of the test). The point is to be able to use the optimization module of pyleecan to optimize other function than pyleecan ones. 
 If you are interested by optimization with pyleecan models please check: https://pyleecan.org/09_tuto_Optimization.html

Best regards,
Pierre