Eomys/pyleecan #625: solve_femm

faraz4714, 2023-05-31:
Dear all,
I have a question and I will be grateful if anyone answers me.
I run my simulation as same as the tutorial but I want to access Magnetic flux density and Magnetic field that are the output of the solve_femm function but in the output of OUTmag there isn't any solve_femm function and it is included MagFEMM and this function requires 11 input that I don't know how i can get these input to them 
can you please someone help me? 
Best regards,
Faraz Mirali

```
rom pyleecan.Classes.MagFEMM import MagFEMM

simu_femm.mag = MagFEMM(
    type_BH_stator=0, # 0 to use the material B(H) curve,
                      # 1 to use linear B(H) curve according to mur_lin,
                      # 2 to enforce infinite permeability (mur_lin =100000)
    type_BH_rotor=0,  # 0 to use the material B(H) curve,
                      # 1 to use linear B(H) curve according to mur_lin,
                      # 2 to enforce infinite permeability (mur_lin =100000)
    file_name = "", # Name of the file to save the FEMM model
    is_fast_draw=True,  # Speed-up drawing of the machine by using lamination periodicity
    is_sliding_band=True,  # True to use the symetry of the lamination to draw the machine faster
    is_calc_torque_energy=True, # True to calculate torque from integration of energy derivate over rotor elements
    T_mag=101,  # Permanent magnet temperature to adapt magnet remanent flux density [°C]
    is_remove_ventS=False,  # True to remove stator ventilation duct
    is_remove_ventR=False,  # True to remove rotor ventilation duct
)

# Only the magnetic module is defined
simu_femm.elec = None
simu_femm.force = None
simu_femm.struct = None

simu_femm.mag.is_periodicity_a=True
simu_femm.mag.is_periodicity_t=True
simu_femm.mag.nb_worker = 4  # Number of FEMM instances to run at the same time (1 by default)
simu_femm.mag.is_get_meshsolution = True # To get FEA mesh for latter post-procesing
simu_femm.mag.is_save_meshsolution_as_file = True # To save FEA results in a dat file

out_femm = simu_femm.run()
simu_femm.mag.solve_FEMM()
```
```
   TypeError                                 Traceback (most recent call last)
Cell In[42], line 3
      1 import matplotlib.pyplot as plt
      2 #plt.plot(out.femm.mag,Ia)
----> 3 simu_femm.mag.solve_FEMM()
      4 # out_femm.mag.axes_dict

TypeError: solve_FEMM() missing 11 required positional arguments: 'femm', 'output', 'out_dict', 'FEMM_dict', 'sym', 'Nt', 'angle', 'Is', 'Ir', 'angle_rotor', and 'is_close_femm'
```

BonneelP, 2023-05-31:
Hello,

solve_FEMM is an internal method of MagFEMM. All the results are available in output.mag after the call to simu.run.
If you want to have access to these results on the mesh, you need to set simu.mag.is_get_meshsolution= True cf https://pyleecan.org/07_tuto_loss.html.

Best regards,
Pierre

faraz4714, 2023-05-31:
finally, I find the answer for this issue and the command is `out_femm.mag.meshsolution.solution[0].field.components["comp_x"].values`
but when I want to plot B versus H, according to this code: 

```
import matplotlib.pyplot as plt
B = sqrt((out_femm.mag.meshsolution.solution[0].field.components["comp_x"].values[:,0])**2 + (out_femm.mag.meshsolution.solution[0].field.components["comp_y"].values[:,0])**2)
H = sqrt((out_femm.mag.meshsolution.solution[1].field.components["comp_x"].values[:,0])**2 + (out_femm.mag.meshsolution.solution[1].field.components["comp_y"].values[:,0])**2)
plt.plot(H,B)
plt.show()
```
I got this graph:
[image]

however, I expect to get something like this
[image]

can you help me for finding a solution, please?

Also, I try to plot Bx versus time but I don't understand the title of it, I mean "indice = 0" What is it?

`out_femm.mag.meshsolution.solution[0].field.plot_2D_Data("time",component_list=["comp_x"])`
[image]

faraz4714, 2023-06-01:
also, I tried this code:
```
group_stator = out_femm.mag.meshsolution.get_group("stator core")
B_s = group_stator.get_field("indice","time","component",label = "B")
H_s = group_stator.get_field("indice","time","component",label = "H")
import matplotlib.pyplot as plt
plt.plot(H_s[0,:,0], B_s[0,:,0])
plt.show
```
but I got this shape
[image]

It seems the general shape is the same but something is wrong 

Best regards,
Faraz Mirali

EmileDvs, 2023-06-01:
Hello Faraz,

For your information Femm does not implement BH curves with hysteresis. If I get it right, you want to look at B(H) over one electrical period at a specific point (node) of the mesh. It seems to me you plot all nodes for all times.

Best regards,
Emile

faraz4714, 2023-06-06:
Dear all,

I am seeking to obtain the hysteresis curve of my magnets and subsequently plot the load line on them, as demagnetization is a crucial parameter in motor design. When attempting to extract the magnetic induction (B) and magnetic field strength (H) values for a specific magnet (e.g., magnet_rotor_0) from Femm, I obtained an array with dimensions (1000, 336, 2). The left-to-right order represents time, number of nodes, and the values of the x and y axes, respectively. However, when I implemented the following code, I encountered a shape that differed significantly from the expected hysteresis curve. I would greatly appreciate any suggestions or insights on how to proceed.

Best regards,
Faraz Mirali
```
import matplotlib.pyplot as plt
import numpy as np
group_magnets = out_femm.mag.meshsolution.get_group("rotor magnets_0")
B = group_magnets.get_field("time","indice","component",label = "B")
H = group_magnets.get_field("time","indice","component",label = "H")
plt.scatter(H[:,0,1],B[:,0,1])
```

BonneelP, 2023-06-09:
Hello,

Sorry for the long time to answer, I'm not an expert on this subject so I needed to investigate. In your code is it normal that you plot only the "Y" component (H[:,0,1]) ?
On my side I tried with the following formula: 
[image]
I defined a linear magnet without B(H) curve so the result should be correct. 

Can you check that your magnet material has a B(H) curve defined and that this B(H) curve is set in the corresponding material in FEMM ? Maybe pyleecan handles only B(H) curves for lamination materials and not for magnets. It should not be hard to correct if it is the case. 

Best regards,
Pierre