Eomys/pyleecan #635: changing the rotor speed doesn't have any effect on the average torque

faraz4714, 2023-07-14:
Dear all,

I have conducted simulations of my IPMSM motor using Pyleecan. However, I encountered an issue when changing the rotor speed. For example, when I increased the rotor speed from 4500 rpm to 450000 rpm (100 times higher), the average torque of the motor remained the same. As you may be aware, the relationship between average torque and rotor speed is expected to be constant up to a certain value and then decrease with further increases in rotor speed until reaching zero.

I would appreciate your assistance in addressing this matter. If needed, I can provide further explanations and details regarding it.

Best regards,
Faraz Mirali

gforti84, 2023-07-14:
Did you change the input current density? If not, torque should be the same.

gforti84, 2023-07-14:
> thank you for your reply
> could you please explain to me how I can obtain the Torque-Speed
> Characteristics(like the picture [below)]
> [image: 3]
> <https://user-images.githubusercontent.com/131860530/253582407-61574f11-75b5-447d-9887-ea44188e18cc.png>

Here is an example for a IPMSM. You must change the operating point (change
Id, fist image, and Iq, second imag):
[image: image.png]
[image: image.png]

faraz4714, 2023-07-14:
thank you for your reply
I am not able to see your images can you upload them again please

gforti84, 2023-07-14:
[image]
[image]

gforti84, 2023-07-14:
These were generating using a different software, MotorCAD.

BonneelP, 2023-07-24:
Hello,

Sorry for the late answer. We have started to add MTPA into pyleecan but it is not fully validated. If you are interested you can investigate how it works with:
pyleecan\Functions\Electrical\comp_MTPA.py
pyleecan\Methods\Simulation\ElecLUTdq\solve_MTPA.py
Tests\Validation\Electrical\test_ElecLUTdq_solve_MTPA.py
Tests\Validation\Electrical\test_MTPA_Stel_2V.py
Tests\Validation\Electrical\test_MTPA_Stel.py

Best regards,
Pierre