Eomys/pyleecan #638: Installation hint: could not bulid wheels for gmsh-sdk, matplotlib

YongYahn, 2023-07-25:
When I install the software on anaconda with the command: pip install pyleecan[full], it says that the libraries gmsh-sdk and matplotlib cannot be installed correctly, but in fact, I already have gmesh 4.11.1 and matplotlib 3.7.2 installed on my anaconda. 

[image]

BonneelP, 2023-07-26:
Hello,

Thank you for your interest in pyleecan. In fact pyleecan requires a version of matplotlib>=3.3.2 and <=3.3.4 which conflict with your current Anaconda version. You can find all the requirements and the corresponding versions here: https://github.com/Eomys/pyleecan/blob/master/requirements-full.txt

To go forward, you can either try a more recent version of matplotlib but we can't say how stable it will be or you can install a new python outside of Anaconda using: https://pyleecan.org/get.pyleecan.html (we used to recommend to use anaconda but we no longer for more requirements flexibility). The GUI installer can also be an option depending on what you are looking for.

We plan to investigate moving Pyleecan to Python 3.11 (today we recommend 3.8), that should remove lot of version requirements. I don't know when we will have time for that. 

Best regards,
Pierre

YongYahn, 2023-07-31:
Thank you very much for your reply, I think the pyleecan platform is very helpful for the in-depth study of motors, and I look forward to the next version will be able to increase the version of python to the latest version, to eliminate the compatibility of some libraries during the installation of the software, thank you!