Eomys/pyleecan #671: Calculation of losses for MachineUD

jeslinjoseph2412, 2023-12-12:
Hello,
I have a user defined machine(Machine_UD9 with sleeeve:I have created a json file by designing the machine in pyleecan.And later i imported the json file to calculate the loss.But i am not able to do the loss calculation and it showing a error as MachineUD' object has no attribute 'rotor'.How i can do the loss calculation of a user defined machine.I have done the loss calculation without sleeve.But the machine was SPISM.Later i changed the SPISM to MachineUD.How i can solve this issue.

BonneelP, 2023-12-14:
Hello,

Thank you for your interest in pyleecan. 
Can you share a screenshot of the machine that you are trying to simulate ? For now MachineUD is not fully compatible with the FEMM coupling (and then the loss models that are computed on the results from FEMM). Depending on the complexity of your machine we should be able to help you to make it work with a MachineUD. 

Best regards,
Pierre

jeslinjoseph2412, 2023-12-14:
Hello,
As the similar way the tutorials in the pyleecan  ,i am doing the loss
calculations. But for the MachineUD.The calculations are not working and
showing error as machine ud does not have rotor attribute. The json file
define the rotor and stator as list.

jeslinjoseph2412, 2023-12-14:
Hello,
Moreover the machine have stator , magnet and sleeve.I am not able to share
the screenshot of machine.