Eomys/pyleecan #681: Symmetry problems

Zelioo, 2024-02-17:
Hello everybody.

We've been using pyleecan in order to plot magnetic field lines of various synchronous reluctance motors using built-in pyleecan models and custom-made materials.

Yet, even when using the default features, we've stumbled on an intriguing matter : The field lines don't look symmetrical at all. Does anyone know if this behavior is normal or if there are ways to improve the 'look' of the field lines.

Best regards.
[image]

Zelioo, 2024-03-01:
I'm sorry if it's a basic mistake from my side but I'm just a beginner with Pyleecan

BonneelP, 2024-03-04:
Hello,

Sorry for the answer delay, I wanted to investigate by simulating the machine on my side but I didn't have the time. My guess would be that the machine has p=2 => 2 pole pairs which means that only a single pole is drawn in FEMM and then I expect FEMM to have set anti-periodic boundary conditions. To check if the simulation is correct, you can remove the symmetry to draw the full machine (simu.mag.is_peridodicity_a = False)

Best regards,
Pierre

crowdnidhi-design, 2024-03-06:
**Dear Pierre,

We tried to update the simu.mag.is_peridodicity_a = False to True in your code example( notebook ) below but it seems it get stuck

look at the error below ( if we skew the rotor does it create problems ?.)**

**
Exception                                 Traceback (most recent call last)
Cell In[5], line 1
----> 1 output_ref = simu_ref.run()

File c:\users\haris\appdata\local\programs\python\python38\lib\site-packages\pyleecan\Methods\Simulation\Simulation\run.py:52, in run(self)
     49     self.init_logger(results)
     51     # Compute the simulation
---> 52     self.run_single()
     54 if self.index is None and self.layer == 0:
     55     msg = "End of simulation"

File c:\users\haris\appdata\local\programs\python\python38\lib\site-packages\pyleecan\Methods\Simulation\Simu1\run_single.py:31, in run_single(self)
     29     self.elec.run()
     30 if self.mag is not None:
---> 31     self.mag.run()
     33 if self.force is not None:
     34     self.force.run()

File c:\users\haris\appdata\local\programs\python\python38\lib\site-packages\pyleecan\Methods\Simulation\Magnetics\run.py:46, in run(self)
     44 if Nslices > 1:
     45     self.get_logger().info("Solving slice 1 / " + str(Nslices))
---> 46 out_dict = self.comp_flux_airgap(output, axes_dict, Is_val=Is_val, Ir_val=Ir_val)
     48 is_loop = True
     50 if Nslices == 1:

File c:\users\haris\appdata\local\programs\python\python38\lib\site-packages\pyleecan\Methods\Simulation\MagFEMM\comp_flux_airgap.py:169, in comp_flux_airgap(self, output, axes_dict, Is_val, Ir_val)
    146     (
    147         B_elem,
    148         H_elem,
   (...)
    165         filename=self.import_file,
    166     )
    167 else:
    168     # Without parallelization
--> 169     B_elem, H_elem, mu_elem, A_node, meshFEMM, groups, A_elem = self.solve_FEMM(
    170         femm,
    171         output,
    172         out_dict,
    173         FEMM_dict=FEMM_dict,
    174         sym=sym,
    175         Nt=Nt,
    176         angle=angle,
    177         Is=Is_val,
    178         Ir=Ir_val,
    179         angle_rotor=angle_rotor,
    180         is_close_femm=self.is_close_femm,
    181         filename=self.import_file,
    182     )
    184 # Store FEMM_dict in to avoid drawing the machine several times
    185 output.mag.internal.FEMM_dict = FEMM_dict

File c:\users\haris\appdata\local\programs\python\python38\lib\site-packages\pyleecan\Methods\Simulation\MagFEMM\solve_FEMM.py:217, in solve_FEMM(self, femm, output, out_dict, FEMM_dict, sym, Nt, angle, Is, Ir, angle_rotor, is_close_femm, filename, start_t, end_t, Nmess)
    214 femm.mi_analyze()
    216 # Load results
--> 217 femm.mi_loadsolution()
    219 # Get the flux result
    220 if is_sliding_band:

File c:\users\haris\appdata\local\programs\python\python38\lib\site-packages\pyleecan\Classes\_FEMMHandler.py:2297, in _FEMMHandler.mi_loadsolution(self)
   2296 def mi_loadsolution(self):
-> 2297     self.callfemm("mi_loadsolution()")

File c:\users\haris\appdata\local\programs\python\python38\lib\site-packages\pyleecan\Classes\_FEMMHandler.py:280, in _FEMMHandler.callfemm(self, myString)
    278 # print(myString) # uncomment to see all the calls made to femm while working
    279 if is_windows_os:
--> 280     x = self.callfemm_Windows(myString)
    281 else:
    282     x = self.callfemm_Linux(myString)

File c:\users\haris\appdata\local\programs\python\python38\lib\site-packages\pyleecan\Classes\_FEMMHandler.py:218, in _FEMMHandler.callfemm_Windows(self, myString)
    216 elif x[0] == "e":
    217     ErrorMsg = x.replace(",", " ").replace("1j", "I")
--> 218     raise Exception(ErrorMsg)
    219 else:
    220     x = eval(x)

Exception: error: Some regions in the problem have been defined
by more than one block label.  These potentially
problematic regions will appear as selected in
the initial view.

Zelioo, 2024-03-09:
Thank you very much,
Setting the parameter to true ( or in my case unticking the angular periodicity in the femm simulation window in the gui ) now allows me to plot, the full machine. Yet, for some reasons when I get the femm file after running the simulation ( instead of getting it from the Drawing Summary window) it seems that the flux density is equal to 0 everywhere.

BonneelP, 2024-03-11:
Great :) I think the reason is that we save the .fem file just after drawing the machine. The current are set in a second step so maybe you have the original fem file without current ?
Normally when running the simulation from the GUI, in the result folder you should find both .fem and .ans with the current defined in the last step of the GUI.

Detest1, 2024-03-11:
but I think we don't get current info please.
We were also confused sometimes.

Detest1, 2024-03-11:
We tried to open the .fem file using FEMM 4.2 and check the circuit properties but did not find the current info

Zelioo, 2024-03-12:
Well, after launching the simulations from the GUI I get a femm file which when opened in FEMM doesn't show any magnetic flux. 
In the other hand, when I open the femm file given before simulation ( and before defining current ), it does show flux density.