Eomys/pyleecan #683: Requested for updated/current. Pyleecan Advanced - Part 1: Optimization

crowdnidhi-design, 2024-02-19:
Dear All,

Please share the updated/current "Pyleecan Advanced - Part 1: Optimization".  The earlier version of the notebook is attached for your reference.
I was not able to find the updated document...
[Pyleecan_Advanced_1_Optimization_undertest.zip](https://github.com/Eomys/pyleecan/files/14325955/Pyleecan_Advanced_1_Optimization_undertest.zip)

BonneelP, 2024-02-22:
Hello,

Thank you for your interest in pyleecan. 

In your other issues (#678 for instance), it seems that you are currently using pyleecan 1.4.2 although the latest release is 1.5.0. Which version of Python are you using ? As explained on the home page "Pyleecan is not yet compatible with Python 3.11 due to PySide2. We plan to update pyleecan to 3.11 when we will have the time to do so." (it seems to create an issue in Pypi that doesn't install the latest version sometimes).

The latest version of the tutorials are available on pyleecan website here: https://pyleecan.org/tutorials.html => https://pyleecan.org/_static/Release/1_5_0/Tutorials.zip. Normally all the tutorials were updated for the release 1.5.0.

Best regards,
Pierre Bonneel

crowdnidhi-design, 2024-02-25:
Well, I have the latest version of Pyleecan but in your updated tutorial we are not able to find options to vary slot dimensions, rotor magnet dimensions, angle, and air gap to run the optimization. Hence we wanted to know if you have the updated book with the options we were interested in.