Eomys/pyleecan #685: back emf constant and emf measurement at different speeds

crowdnidhi-design, 2024-03-06:
Dear All,

I was wondering if we have a back emf constant measurement routine and the equation it uses.

Detest1, 2024-04-08:
Guys any update

carno44, 2024-09-04:
Hi,

I want to plot the Torque vs Speed curve of the motor. I used the Tutorial -8 which used Toyota Prius 2004 machine and used only mag module. I added following two lines to compute and plot the emf. 

`output_ref.mag.comp_emf() #To compute the back emf (V)
output_ref.mag.emf.plot_2D_Data("time")`

But how to use this emf to calculate the net torque for the speed? Can it be done directly in multi-sim? IN Tutorial-8 MultiSim, right now the speed is held constant as varying it will not change the Torque value. 

Best regards,
Carno