Eomys/pyleecan #687: Pyleecan is throwing lot of errors

Detest1, 2024-04-04:
Hi All,

Can anyone let me know what is wrong with the attached file.
Off lately any model we are building is throwing lot of errors.[
[ex_poc.json](https://github.com/Eomys/pyleecan/files/14862113/ex_poc.json)
[ex_poc.json](https://github.com/Eomys/pyleecan/files/14862117/ex_poc.json)

](url)

Detest1, 2024-04-04:
[ex_poc.json](https://github.com/Eomys/pyleecan/files/14862345/ex_poc.json)

Detest1, 2024-04-04:
I tried to generate a FEMM file and opened it in FEMM. It is throwing error too.

Detest1, 2024-04-04:
Does Pyleecan recognise 10P/12S machine ?

Detest1, 2024-04-08:
Dear Pierre,

We tried to run via GUI and Script ( as explanied in your tutorial)but it throws an error. Let me share the error pic for your reference.
I have shared the entire machine file for your reference. You can just import into your installation and run. I believe it will throw error...
May be something wrong with the stator-slot type we have chosen?.

please note we tried to build a few other models and they are working 100% as expected.

BonneelP, 2024-04-08:
Hello, 

I didn't had time to study your json file as my schedule is rather constraint right now. With few explanation/screenshot from your side it would be less time consuming on my side and I may be able to answer quickly your issue. I will see if I find some time to check your json.

Best regards,
Pierre