Eomys/pyleecan #701: 09 - How to solve optimization problem in Pyleecan not useable

wowcodeisgood, 2024-06-20:
i have many questions for pyleecan and i can go to no one whether documentation or youtube, there is no good explanantion of the Methods Classes etc. and if Tutorials exist there are generally false Methods for Example in the Opimization, we vary geometrical variables without changing the winding so fill factors up to 180% are easily possible.I want set up this right to have something useable. And I do not understand how Objectives like efficiency etc. can be put in the simulation. And then there is the point of how i can add new geometries for me and maybe share it later with the community. There is no useful Documentation. Maybe anyone can help me:(

BonneelP, 2024-06-21:
Hello,

If I understand correctly you want to run an optimization simulation that would change the geometry of a slot while keeping a constant fill factor and add an objective on efficiency. Regarding the first one, there is no function in pyleecan to do that, you will need to write yourself the setter function of the DesignVariable (a function that takes the simulation and a value in argument, the value would be the desired slot dimensions and the function would edit several parameters to adapat the slot and the winding). For the efficiency, it's still a work in progress in pyleecan there is a notebook [tuto_Efficiency_map.ipynb](https://github.com/Eomys/pyleecan/blob/master/Tutorials/tuto_Efficiency_map.ipynb) but we will need to check if this is still relevant. 

Best regards,
Pierre

wowcodeisgood, 2024-06-21:
Hello Bonnel,
Thank you for your support you are great.
The First issue could it look like this:

def set_slot_geometry(simulation, slot_dimensions):
    """
    Setter function to adjust the slot dimensions and winding parameters
    while keeping the fill factor constant.
    
    Parameters:
    - simulation: The simulation object from Pyleecan
    - slot_dimensions: The new slot dimensions to be set
    """
    # Extract current winding parameters
    winding = simulation.machine.stator.winding

    # Update slot dimensions
    simulation.machine.stator.slot.H0 = slot_dimensions['H0']
    simulation.machine.stator.slot.H1 = slot_dimensions['H1']
    simulation.machine.stator.slot.H2 = slot_dimensions['H2']
    simulation.machine.stator.slot.W0 = slot_dimensions['W0']
    simulation.machine.stator.slot.W1 = slot_dimensions['W1']
    simulation.machine.stator.slot.W2 = slot_dimensions['W2']

    # Compute new slot area
    slot_area = compute_slot_area(simulation.machine.stator.slot)

    # Update winding parameters to keep fill factor constant
    fill_factor = winding.fill_factor
    conductor_area = fill_factor * slot_area / winding.Npcpp
    winding.conductor.area = conductor_area

    # Ensure other necessary parameters are adjusted accordingly
    # (e.g., number of turns, layers, etc.)

def compute_slot_area(slot):
    """
    Compute the area of the slot based on its dimensions.
    
    Parameters:
    - slot: Slot object with dimensions H0, H1, H2, W0, W1, W2
    
    Returns:
    - slot_area: The computed slot area
    """
    # Implement the computation of the slot area based on its dimensions
    # This is a simplified example; adjust as necessary for your specific slot type
    slot_area = slot.H2 * (slot.W0 + slot.W1) / 2
    return slot_area 

Or am I on the false route🫡
Thanks again 
Best regards,

BonneelP, 2024-06-21:
Hello,

You can take inspiration from https://github.com/Eomys/pyleecan/blob/master/Tests/Validation/Multisimulation/test_slot_scale.py
Your setter is linked to a Design Variable, so I would suggest setting a float in an interval for the value rather than a dictionary

The slot area is already available with simulation.machine.stator.slot.comp_surface() and even comp_active_surface(). You can also use simulation.machine.stator. comp_fill_factor() 

To make sure that the fill factor doesn’t change you can either play on the conductor dimensions or introduce a clever scaling rule on your slot that keeps the active surface constant.

Best regards,
Pierre

wowcodeisgood, 2024-07-14:
Hello and Thank you Bonnel, Great work you do! For me this should be
implemented in Library that the winding factor is not allowed to change and
the conductor diameter is the same, and only the Surface has changed you
got more winding so more inductivity and and Impedance in general I think
everyone who want to make a optimization for a electrical machine expects
that this feature is in it because otherwise the simulation makes no sense
because the optimizer is optimizing some thing that is not possible for
example you optimize for Tem_av so you allow to vary the geometry but you
are have not always the highest possible winding in the simulation so you
get not the best machine out of it. Lg Lukas