Eomys/pyleecan #702: Electrostatic and linear machines

nkbw33, 2024-07-05:
Hi!
If electrostatic and linear machines are not already implemented, which as far as I can tell they are not, what would be needed to implement these?

BonneelP, 2024-07-08:
Hello,

I'm not familiar with these machines (I'm not an electrical engineer). Can you share some screenshot/schematics/particularity that needs to be addressed to simulated these kind of machines ? With some information I should be able to guide you in the development to add them. 

Best regards,
Pierre

nkbw33, 2024-07-08:
Hello,

This is an electrostatic machine:
https://ieeexplore.ieee.org/document/8470983
[image]
[image]

This is a linear machine:
[image]
[image]

Best,
Noam

BonneelP, 2024-07-09:
Hello,

Thank you for sharing these information. For now pyleecan is dedicated to radial flux machines, I think we can reuse some existing parts to add both these machines but it would require to find the correct parametrization, create the corresponding objects and add/adapt the models to simulate them so a rather large work I would say. I can give some guidelines and help you on the way but I won't have the time to directly work on the subject I think.

Best regards,
Pierre