Eomys/pyleecan #723: Is the Pyleecan Project Still Actively Maintained?

zmladen, 2025-01-28:
Hello everyone,

I was wondering if the Pyleecan project is still actively maintained and supported? I have noticed that there haven’t been any responses to issues or discussions recently!

I think it is a great tool, and I’d like to know if there are still contributors or maintainers actively involved in its development.

Thanks in advance for any information!

SebGue, 2025-02-10:
Hello zmladen,

thats a good question. I was also wondering if there is still active development by Eomsy. Maybe @BonneelP could briefly tell something about Eomys/Dassault Systems future development plans to us.

From my side, I still plan using Pyleecan and also got some uncommited changes/improvements (especially regarding MagElmer and GMSH model creation). Further, I also plan implementing some at least minimal thermal modeling. Nevertheless, I don't have too much time to contribute at the moment and the near future. 

Since you asked this question, what are your expectations regarding development, needed features, software compatibility etc. ?

Best regards, Sebastian

zmladen, 2025-02-20:
I've created some wire visualizations within the winding active area in my fork (you can follow in discussions). I wanted to see if they are interesting to the community and if they can be merged with the main branch. I was also wondering if Pyleecan could be adapted to support hairpin winding layouts. I'm not sure if SWAT-EM, which is used under the hood, can be applied for that purpose.

BonneelP, 2025-03-06:
Hello all,

Sadly, I have less and less time (professionally and personally) to dedicate to pyleecan so it is hard for me to maintain the project and I'm looking for someone to take the lead to push forward pyleecan. I’m now a father which make finding some time for the project more complicated. We are preparing on official communication from 3DS about our plan with pyleecan following Eomys acquisition but it will take some time to prepare. 

Best regards,
Pierre Bonneel

SebGue, 2025-08-01:
Hello @BonneelP ,

first of all congratulations on becoming a father! That’s a wonderful, exciting and also challenging change — wishing you and your family all the best.

Regarding Pyleecan, I wanted to kindly ask if there is already any update or a rough timeline regarding the planned official communication from 3DS following the Eomys acquisition. Even a short interim statement would be greatly appreciated to help us better understand the current direction and how we might support or align going forward with pyleecan.

Looking forward to hearing from you.

Best regards,
Sebastian